NorthernMan54/rtl_433_ESP #113: How to decode RAW?

tablatronix, 2023-10-29:
### Current Situation

What is this format?
`+62755-451(-35)+1155-456(-36)+1187-436(-37)`...

I looked at some bit decoders and this did not stand out as one of the formats..

I know I saw it doc somewhere, but cannot find it

### Logs

```markdown
{"model":"Generic-Remote","id":57567,"cmd":1,"tristate":"1X001Z11000Z","protocol":"Generic Remote SC226x EV1527","rssi":-34,"duration":173002}
```

`rtl_433_ESP(6): RAW (787970): +62755-451(-35)+1155-456(-36)+1187-436(-37)+385-1238(-37)....`
```

### Configuration

```json
.
```

### Environment

OMG cc1101 decoder

### Process Supervisor

not applicable

NorthernMan54, 2023-10-30:
It is a raw representation of the pulse pattern being received.  If you can imagine a pulse stream, and the + number is the width in MS of the up pulse, and the -number is the width in MS of the down pulse.  And in brackets is the signal strength.

```
rtl_433_ESP(6): RAW (787970): +62755-451(-35)+1155-456(-36)+1187-436(-37)+385-1238(-37)....
```

This is probably 110 in binary, depending on the modulation scheme and device format

tablatronix, 2023-10-31:
Ah ok thanks, do you know how to get this into a standard format like rtl and sdrs use .ook .cu8 etc? Is there a pipeline for that?

NorthernMan54, 2023-10-31:
Try this directive, it will run the rtl_433 analyzer in the code base.

`RTL_ANALYZER`

cgmckeever-orbis, 2024-06-01:
@tablatronix did you ever figure some of this out?
If you have the tristate, you can transmit that code via [RC-Switch ](https://github.com/sui77/rc-switch/blob/master/examples/SendDemo/SendDemo.ino#L51) 

Im just trying to figure out how to get those codes ...

closed 2024-11-30 by NorthernMan54

tablatronix, 2024-11-30:
I never did figure out how to convert formats directly, I think i gave up debugging this and just found something that could record it and work and bought a broadlink rmp