NorthernMan54/rtl_433_ESP #114: 2 different frequencies with lilygo-rtl_433

Dubi256, 2023-11-06:
### Current Situation

Hello,

I have two devices:
"TFA Marbella Pool Thermometer" works well with bulild_flag:
DCC1101_FREQUENCY=868.00

"Bresser Weather Center 5-in-1" works well with bulild_flag:
DCC1101_FREQUENCY=868.30

But both together do not work.
What do I have to set so that they work at the same time?

with the original merbanan/rtl_433 both works at the same time with a RTL2832U on a Rasp.
[image]

Thanks!
Markus

### Logs

```markdown
-
```

### Configuration

```json
build_flags =
  ${com-esp32.build_flags}
  '-UZmqttDiscovery'          ; disables MQTT Discovery
  '-DvalueAsATopic=true'    ; MQTT topic includes model and device
  '-DGateway_Name="OMG_lilygo_rtl_433_ESP"'
  '-DZgatewayRTL_433="rtl_433"'
  '-DZradioSX127x="SX127x"'
  '-DZdisplaySSD1306="LilyGo_SSD1306"'
  '-DDISPLAY_IDLE_LOGO=false'
  '-DOOK_MODULATION=false'
  '-DCC1101_FREQUENCY=868.00'
```

### Environment

- OS: Windows10

### Process Supervisor

not applicable

NorthernMan54, 2023-11-06:
Experiment with the setting and maybe one  will expand receiver frequency bandwidth Two boards, they are cheap?Write a timer routine to change frequency every few minutes?Run it inside OMG, and change frequencies every few minutes?

Configuration

Dubi256, 2023-11-07:
Hello NorthernMan54,

First of all, thank you very much for your support!

I have tested all frequencies between 868.00 and 868.30, but unfortunately both devices did not work simultaneously at any frequency.

Do you have any tips on which settings I could change on the receiver?
Why can I receive both devices at the same time with the USB-stick and not with the Lilygo?

two devices are of course a possibility....

Thanks!

NorthernMan54, 2023-11-07:
@Dubi256 If you look at the data sheet for the transceiver module, they have a lot of settings.  https://cdn-shop.adafruit.com/product-files/3179/sx1276_77_78_79.pdf

And they are set here in the code - https://github.com/NorthernMan54/rtl_433_ESP/blob/3fea1cf678212ea5ef70e38f625bc8505f73bb28/src/rtl_433_ESP.cpp#L235

Please keep in mind that the LilyGo and the USB-stick are two totally different technologies, and the signal reception logic between the this package and rtl_433 is totally different.  rtl_433 uses a software defined signal demodulator, where this package leverages the signal demodulator with the receiver chipset.

habub, 2023-11-23:
The USB sticks have very powerfull receivers designed originally for receiving digital TV (DVB-T). Combined with a PC you can receive and view a wide range of frequencies. I think up to 2 Mhz.
Radio receivers are dedicated for a much smaller bandwidth. I'm not sure your 300 khz is within their capabilities. You need to check the specs and maybe test some register modifications.

1technophile, 2023-11-23:
You could implement a routine change of frequency for the receiver. Depending on your emitters frequency sending, 5 min at each frequency, for example

closed 2024-11-30 by NorthernMan54