NorthernMan54/rtl_433_ESP #117: RTL_ANALYZER

habub, 2023-11-16:
### Current Situation

Hello,
Regarding the analyzer call in signalDecoder.cpp:
```
#ifdef RTL_ANALYZER
      pulse_analyzer(rtl_pulses, rtl_433_ESP::ookModulation ? 1 : 2);
#endif
```
I didn't find any reference to RTL_ANALYZER for the esp32_cc1101 environment.
Does this work?

### Logs

```markdown
N/A
```

### Configuration

```json
N/A
```

### Environment

- OS:
- Software:
- Node:
- npm:

### Process Supervisor

not applicable

NorthernMan54, 2023-11-16:
Is your format correct ?

`-DRTL_ANALYZER=true`

Am thinking your missing the -

habub, 2023-11-16:
Nevermind, I found the typo. Sorry for the inconvenience...
I will update how it goes.

habub, 2023-11-16:
I see it's working, but you don't output the bitbuffer bits if I understand correctly.
Is this on purpose? Even when a decoding succeeded.

closed 2023-11-23 by habub