NorthernMan54/rtl_433_ESP #123: SX1278 with UART instead of SPI

kobold81, 2023-12-15:
### Current Situation

Hello,

I bought by accident an UART board with the SX1278 (see link below).
Do you plan to add a UART mode to your code?

https://de.aliexpress.com/item/32791728376.html?aff_fcid=e01092b5ea044ecc96d6b47f2c0f5621-1702667469461-01142-eOHotRkU&aff_fsk=eOHotRkU&aff_platform=product&sk=eOHotRkU&aff_trace_key=e01092b5ea044ecc96d6b47f2c0f5621-1702667469461-01142-eOHotRkU&terminal_id=52986ac158f3448986309ed6a3ee46ba&gatewayAdapt=glo2deu

### Proposed Change

Adding UART if possible

NorthernMan54, 2023-12-15:
I looked at that unit, and does it have an ESP32 ?

kobold81, 2023-12-16:
No, that module is only the sx1278. Finding that unit with 868mhz and spi was impossible. the esp32 would be connected using uart.
Sx1278 seems to be plenty around on ali in 433mhz but not 868mhz (for bresser weather device)

NorthernMan54, 2023-12-16:
Without the esp32 being directly connected then, this will not work.

if you want to play around a bit, the core of the signal receiver logic watches a gdo pin on the sx1278 to receive the signal. If you can figure that out it might work.

take a look at the cc1101 wiring for inspiration

closed 2024-11-30 by NorthernMan54