NorthernMan54/rtl_433_ESP #128: heltec-rtl_433-fsk setup for Bresser 7 in 1 weatherstation

2nice4u, 2024-02-01:
### Current Situation

With the original rtl_433 version and an SDR stick I can retrieve the BRESSER 7-in-1 weather station with the following command: rtl_433 -f 868.2M -s 1000k -X 'n=Bresser-7in1,m=FSK_PCM,s=125,l=125,r= 9000,preamble=2dd4'
But how can I do this in OpenMQTTGateway v1.7.0? How can I control rtl_433 there or is there a config file?

### Logs

```markdown
time      : 2024-02-02 00:02:44
model     : Bresser-7in1 count     : 1             num_rows  : 1             rows      :
len       : 271          data      : ddb73720b2aa10aaaaaaaaaaa8febceac8aaaaaaaaaaaaaaaa000000000000000000
codes     : {271}ddb73720b2aa10aaaaaaaaaaa8febceac8aaaaaaaaaaaaaaaa000000000000000000
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2024-02-02 00:02:44
model     : Bresser-7in1 id        : 40330
Startup   : 1            Temperature: 16.4 C       Humidity  : 62            Wind Gust : 0.0 m/s       Wind Speed: 0.0 m/s
Direction : 180          Rain      : 0.2 mm        Light     : 0.000 klx     Light     : 0.000 lux     UV Index  : 0.0
Battery   : 1            Integrity : CRC
```

### Configuration

```json
rtl_433 -f 868.2M -s 1000k -X 'n=Bresser-7in1,m=FSK_PCM,s=125,l=125,r= 9000,preamble=2dd4'

I need a way to execute this command in OpenMQTTGateway.
```

### Environment

- OS: 
- Software: OpenMQTTGateway v1.7.0
- Node:
- npm:

### Process Supervisor

hb-service

NorthernMan54, 2024-02-02:
Rtl_433_esp is controlled thru compiler directives and they are listed in the README.

in your example, you are using a Flex Decoder which is not supported due to resource constraints on the esp32.  But you could try setting the frequency and enabling FSK signal demodulation and see what is received.

2nice4u, 2024-02-23:
Unfortunately the protocol: "[**173] Bresser Weather Center 7-in-1, Air Quality PM2.5/PM10 7009970, CO2 7009977, HCHO/VOC 7009978 sensors**" is not supported by rtl_433_esp.

rtl_433 -f 868.3M -s 1000k -R 173

NorthernMan54, 2024-11-30:
I refreshed the device decoders from rtl_433 24.10, maybe this is resolved?