NorthernMan54/rtl_433_ESP #130: LCD Display is blank on "LILYGO® LoRa32 V2.1 ESP32 LoRa Development Board" (SX1278 433MHz)

puterboy, 2024-03-14:
### Current Situation

I used PlatformIO to install the "OOK_Receiver" example on my LILYGO® LoRa32 V2.1 ESP32 (SX178 433MHZ) development board.
The serial log shows that it is properly getting signals from my various temperature sensors.
However, the built-in LCD display is blank whereas I had expected it perhaps to display the sensors being found.

Is that to be expected? or is there something broken with either my board or my installation?

Thanks!

### Logs

```markdown
None
```

### Configuration

```json
PlatformIO/Visual Studio running under Win11 on LILYGO® LoRa32 V2.1 ESP32 LoRa Development Board.
```

### Environment

Windows 11

### Process Supervisor

not applicable

NorthernMan54, 2024-03-14:
I guess the picture in the README is a little bit misleading, as it shows the board running OpenMQTTGateway, which includes this library.  OpenMQTTGateway also includes the display code as well.

To enable the display, please install openmqttgateway

puterboy, 2024-03-14:
OK - so you are saying install OpenMQTTGateway **INSTEAD** of OOK_RECEIVER?

puterboy, 2024-03-14:
Thanks! That works!

closed 2024-03-14 by puterboy