NorthernMan54/rtl_433_ESP #134: CC1101 goes deaf for exactly 1 hour every few hours

melyux, 2024-03-30:
### Analysis

[image]

I'm using OpenMQTTGateway to utilize this library and I got it working perfectly with my two FSK sensors on 433MHz. But every few hours, for exactly an hour, the ESP32 running OMG will stop getting data from both sensors for exactly one hour. The intervals between the deafness vary, but the deafness length is always 1 hour.

I'm assuming this caused by an internal check in this library to reset the CC1101 when it goes deaf? But yeah that's happening.

Any suggestions? I'm open to ditching the CC1101 if it's [hopeless](https://github.com/NorthernMan54/rtl_433_ESP/issues/16#issuecomment-909328527) and switching to an equivalent radio, but not sure which one.

### Expected Behavior

The library decodes data from the devices without any intervals of no data.

### Steps To Reproduce

1. Set up the latest-commit OpenMQTTGateway with RTL_433, using the default build `env:esp32dev-rtl_433` but with these changed options:
* OOK_MODULATION=false
* RF_MODULE_CS=5
2. Use FSK devices and see that they stop receiving for 1 hour exactly anywhere from a few minutes to several hours later, and again after that, and so on.

### Logs

```markdown
Trace OMG messages at the time of going deaf:

T: isAdupl?
N: Send on /RTL_433toMQTT/Fineoffset-WN34/12333 msg {"model":"Fineoffset-WN34","id":12333,"battery_ok":0.5,"battery_mV":1380,"temperature_C":19.3,"mic":"CRC","protocol":"Fine Offset Electronics WN34 temperature sensor","rssi":-34,"duration":7000}
T: simplePub - ON
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RTL_433toMQTT/Fineoffset-WN34/12333/model msg: Fineoffset-WN34 
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RTL_433toMQTT/Fineoffset-WN34/12333/id msg: 12333 
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RTL_433toMQTT/Fineoffset-WN34/12333/battery_ok msg: 0.5 
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RTL_433toMQTT/Fineoffset-WN34/12333/battery_mV msg: 1380 
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RTL_433toMQTT/Fineoffset-WN34/12333/temperature_C msg: 19.3 
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RTL_433toMQTT/Fineoffset-WN34/12333/mic msg: CRC 
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RTL_433toMQTT/Fineoffset-WN34/12333/protocol msg: Fine Offset Electronics WN34 temperature sensor 
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RTL_433toMQTT/Fineoffset-WN34/12333/rssi msg: -34 
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RTL_433toMQTT/Fineoffset-WN34/12333/duration msg: 7000 
T: Min ind: 0
T: store code : 12352 / 43185356
T: Col: val/timestamp
T: mem code : 12352 / 43185356
T: mem code : 12352 / 42800323
T: mem code : 237 / 42848905
T: mem code : 12352 / 42877355
T: mem code : 237 / 42912886
T: mem code : 12352 / 42954335
T: mem code : 237 / 42976904
T: mem code : 12352 / 43031368
T: mem code : 237 / 43040890
T: mem code : 237 / 43104921
T: mem code : 12352 / 43108373
T: mem code : 237 / 43168902

T: Enqueue JSON
T: Queue length: 1
N: Send on /RFtoMQTT msg {"active":3,"frequency":433.92,"rssithreshold":-96,"rssi":-106,"avgrssi":-105,"count":12857}
T: simplePub - ON
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RFtoMQTT/active msg: 3 
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RFtoMQTT/frequency msg: 433.9 
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RFtoMQTT/rssithreshold msg: -96 
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RFtoMQTT/rssi msg: -106 
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RFtoMQTT/avgrssi msg: -105 
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/RFtoMQTT/count msg: 12857 
T: Enqueue JSON
T: Queue length: 2
T: Dequeue JSON
N: Send on /SYStoMQTT msg {"uptime":43208,"version":"version_tag","env":"xxx-xxxxxx-xxxx","freemem":132120,"mqttp":"1883","mqtts":false,"msgprc":720,"msgblck":0,"maxq":2,"minmem":50536,"tempc":53.33,"freestck":2236,"eth":false,"rssi":-52,"SSID":"XXxxXXxx","BSSID":"xx:xx:xx:xx:xx:xx","ip":"xxx.xxx.xxx.xxx","mac":"xx:xx:xx:xx:xx:xx","modules":["WebUI","RF","RF2","RTL_433"]}
T: simplePub - ON
T: [ OMG->MQTT ] topic: rf/omg-bay-cc1101/SYStoMQTT/uptime msg: 43208 
[... 172 lines cut ...]
  '-Dwifi_ssid="xxxxxxx"'
  '-Dwifi_password="xxxxx"'
  '-DMQTT_USER="xxxxxxx"'
  '-DMQTT_PASS="xxxxxxx"'
  '-DMQTT_SERVER="xxx.xxx.xxx.xxx"'
  '-DjsonPublishing=false'            ; Disable JSON (separate topics for each thing)
  '-DsimplePublishing=true'           ; Disable JSON (separate topics for each thing)
  '-UZmqttDiscovery'                  ; Disable HA discovery (was enabled in com-esp)
  '-DLOG_LEVEL=LOG_LEVEL_TRACE'      ; Can be set to LOG_LEVEL_TRACE (everything) or LOG_LEVEL_ERROR (only errors)
; *** rtl_433_ESP Options ***
  '-DOOK_MODULATION=false'            ; FSK modulation activated (and ASK modulation disabled)
  '-DRF_MODULE_CS=5'                  ; pin to be used as chip select (wasn't set in esp32dev-rtl_433)
  '-UPUBLISH_UNPARSED'                ; don't publish unparsed signal details (was enabled in esp32dev-rtl_433)
  '-UDEMOD_DEBUG=false'               ; don't display signal debug info (was enabled in esp32dev-rtl_433)
monitor_speed = 115200
```

### Environment

- OS: Mac
- Software: Platformio

### Process Supervisor

not applicable

melyux, 2024-03-30:
Actually, I see that OMG's `env:esp32dev-rtl_433` build has `'-DNO_DEAF_WORKAROUND=true'`. I guess I didn't see the "NO_" there and couldn't conceive why they would turn that workaround off by default in the main RTL_433_ESP build. I'll test with that unset as `'-UNO_DEAF_WORKAROUND'` and see how it goes.

melyux, 2024-03-31:
Never mind, still happens with that unset too.

```
N: Send on /RFtoMQTT msg {"active":3,"frequency":433.92,"rssithreshold":-92,"rssi":-101,"avgrssi":-101,"count":8985}
N: Send on /SYStoMQTT msg {"uptime":26896,"version":"version_tag","env":"bay-cc1101-usb","freemem":135272,"mqttp":"1883","mqtts":false,"msgprc":450,"msgblck":0,"maxq":2,"minmem":52820,"tempc":52.78,"freestck":1900,"eth":false,"rssi":-53,"SSID":"HyperNet","BSSID":"08:62:66:97:A3:28","ip":"192.168.1.66","mac":"40:22:D8:ED:56:EC","modules":["WebUI","RF","RF2","RTL_433"]}
N: Send on /WebUItoMQTT msg {"displayMetric":true,"webUISecure":true,"displayQueue":0}
N: Send on /RFtoMQTT msg {"active":3,"frequency":433.92,"rssithreshold":-92,"rssi":-101,"avgrssi":-101,"count":8985}
N: Send on /SYStoMQTT msg {"uptime":27016,"version":"version_tag","env":"bay-cc1101-usb","freemem":135272,"mqttp":"1883","mqtts":false,"msgprc":452,"msgblck":0,"maxq":2,"minmem":52820,"tempc":52.78,"freestck":1900,"eth":false,"rssi":-55,"SSID":"HyperNet","BSSID":"08:62:66:97:A3:28","ip":"192.168.1.66","mac":"40:22:D8:ED:56:EC","modules":["WebUI","RF","RF2","RTL_433"]}
N: Send on /WebUItoMQTT msg {"displayMetric":true,"webUISecure":true,"displayQueue":0}
N: Send on /RFtoMQTT msg {"active":3,"frequency":433.92,"rssithreshold":-92,"rssi":-101,"avgrssi":-101,"count":8985}
N: Send on /SYStoMQTT msg {"uptime":27136,"version":"version_tag","env":"bay-cc1101-usb","freemem":135272,"mqttp":"1883","mqtts":false,"msgprc":454,"msgblck":0,"maxq":2,"minmem":52820,"tempc":52.78,"freestck":1900,"eth":false,"rssi":-54,"SSID":"HyperNet","BSSID":"08:62:66:97:A3:28","ip":"192.168.1.66","mac":"40:22:D8:ED:56:EC","modules":["WebUI","RF","RF2","RTL_433"]}
N: Send on /WebUItoMQTT msg {"displayMetric":true,"webUISecure":true,"displayQueue":0}
N: Send on /RFtoMQTT msg {"active":3,"frequency":433.92,"rssithreshold":-92,"rssi":-101,"avgrssi":-101,"count":8985}
N: Send on /SYStoMQTT msg {"uptime":27256,"version":"version_tag","env":"bay-cc1101-usb","freemem":135272,"mqttp":"1883","mqtts":false,"msgprc":456,"msgblck":0,"maxq":2,"minmem":52820,"tempc":52.78,"freestck":1900,"eth":false,"rssi":-53,"SSID":"HyperNet","BSSID":"08:62:66:97:A3:28","ip":"192.168.1.66","mac":"40:22:D8:ED:56:EC","modules":["WebUI","RF","RF2","RTL_433"]}
N: Send on /WebUItoMQTT msg {"displayMetric":true,"webUISecure":true,"displayQueue":0}
N: Send on /RFtoMQTT msg {"active":3,"frequency":433.92,"rssithreshold":-92,"rssi":-101,"avgrssi":-101,"count":8985}
N: Send on /SYStoMQTT msg {"uptime":27376,"version":"version_tag","env":"bay-cc1101-usb","freemem":135272,"mqttp":"1883","mqtts":false,"msgprc":458,"msgblck":0,"maxq":2,"minmem":52820,"tempc":52.78,"freestck":1900,"eth":false,"rssi":-55,"SSID":"HyperNet","BSSID":"08:62:66:97:A3:28","ip":"192.168.1.66","mac":"40:22:D8:ED:56:EC","modules":["WebUI","RF","RF2","RTL_433"]}
N: Send on /WebUItoMQTT msg {"displayMetric":true,"webUISecure":true,"displayQueue":0}
N: Send on /RFtoMQTT msg {"active":3,"frequency":433.92,"rssithreshold":-92,"rssi":-101,"avgrssi":-101,"count":8985}
N: Send on /SYStoMQTT msg {"uptime":27496,"version":"version_tag","env":"bay-cc1101-usb","freemem":135272,"mqttp":"1883","mqtts":false,"msgprc":460,"msgblck":0,"maxq":2,"minmem":52820,"tempc":53.33,"freestck":1900,"eth":false,"rssi":-54,"SSID":"HyperNet","BSSID":"08:62:66:97:A3:28","ip":"192.168.1.66","mac":"40:22:D8:ED:56:EC","modules":["WebUI","RF","RF2","RTL_433"]}
N: Send on /WebUItoMQTT msg {"displayMetric":true,"webUISecure":true,"displayQueue":0}
```

NorthernMan54, 2024-03-31:
To see if the deaf workaround is truly the issue, what about adding a logprint around here ?

https://github.com/NorthernMan54/rtl_433_ESP/blob/276bc2cc6cef73037aaf3325b69be99ccd139742/src/rtl_433_ESP.cpp#L447

melyux, 2024-04-09:
Still going... I did erase the flash on the ESP, so no idea what happened. So far so good. Will close unless it happens again

melyux, 2025-03-02:
Never mind, it's happening again on the latest OMG. This time it's not always 1 hour, and it doesn't seem to come back even after restarting the ESP32. Maybe the NO_DEAF_WORKAROUND is on. I'll try the logprint, but by the time I notice it happening, the web UI console log has already scrolled past the time.