NorthernMan54/rtl_433_ESP #135: [43]  Watchman Sonic / Apollo Ultrasonic / Beckett Rocket oil tank monitor

kyuzumaki, 2024-04-15:
### Current Situation

Hi,
I've been using rtl_433 to read an oil sensor. After setting up OpenMQTT gateway to try to replace the SDR system I've discovered it dosent seem to be picking up my tank sensor.
The offical rtl_433 library has this decoder 
[43]  Watchman Sonic / Apollo Ultrasonic / Beckett Rocket oil tank monitor
Is there any way to get this one working?
I would be happy having it ignore every other protocol if that frees up enough resource on the esp to decode just that signal.

Currently setup is a Lilygo_433 with OpenMQTT running and connected to home assistant. Ive let it run for a good while and have not seen any new sensors show up in Home assistant.

### Proposed Change

Add 
[43]  Watchman Sonic / Apollo Ultrasonic / Beckett Rocket oil tank monitor
from trl_433 library

DigiH, 2024-04-16:
> Edit: Ok I'm wrong it's listed under a different number
> Registering protocol [54] "Watchman Sonic / Apollo Ultrasonic / Beckett Rocket oil tank monitor"

When you found the existing decoder being actually included, did you see that it is under the **FSK** section, so the **FSK** modulation will need to be set, as described in the Compile Definition Options section

https://github.com/NorthernMan54/rtl_433_ESP?tab=readme-ov-file#compile-definition-options

Let us know how you get on.

kyuzumaki, 2024-04-16:
Thank you! I noticed it was in a different section but failed to recognise my mistake. Working perfectly now

closed 2024-04-16 by kyuzumaki