NorthernMan54/rtl_433_ESP #136: Wireless M-Bus - How to set the sample rate?

eporsche, 2024-04-16:
### Current Situation

In the descirption the -s flag is set for rtl_433  -> Wireless M-Bus Mode C&T, 100kbps (-f 868.95M -s 1200k) 
However I am not sure how to set it for the rtl_433_ESP - is there a build flag I can use for this purpose?

### Logs

```markdown
N: Send on /RTL_433toMQTT/undecoded signal msg {"model":"undecoded signal","protocol":"signal parsing failed","duration":1996,"rssi":-98,"pulses":18}
```

### Configuration

```json
[env:esp32dev-multi_receiver-pilight-rtl_433_ESP_MBus]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp32.lib_deps}
  ${libraries.rc-switch}
  ${libraries.smartrc-cc1101-driver-lib}
  ${libraries.esppilight}
  ${libraries.newremoteswitch}
  ${libraries.irremoteesp}
  ${libraries.rtl_433_ESP}
build_flags =
  ${com-esp32.build_flags}
  '-DZgatewayIR="IR"'
  '-DZgatewayRF="RF"'
  '-DZgatewayRF2="RF2"'
  '-DZgatewayRTL_433="RTL_433"'
  '-DZgatewayPilight="Pilight"'
  '-DZradioCC1101="CC1101"'
  '-DGateway_Name="OpenMQTTGateway"'
  '-DvalueAsATopic=true'  ; MQTT topic includes model and device (rtl_433) or protocol and id (RF and PiLight)
; *** IR Module Options ***
  '-DIR_EMITTER_GPIO=14'
  '-DIR_RECEIVER_GPIO=25'
; *** RF Module Options ***
  '-DRF_CC1101="CC1101"'  ; CC1101 Transceiver Module
  '-DRF_MODULE_CS=5'      ; pin to be used as chip select
  '-DRF_MODULE_GDO0=12'   ; CC1101 pin GDO0
  '-DRF_MODULE_GDO2=27'   ; CC1101 pin GDO2
;  '-DRF_MODULE_INIT_STATUS=true'    ; Display transceiver config during startup
; *** rtl_433_ESP Options ***
;  '-DRTL_ANALYZER=true'
  '-DOOK_MODULATION=false' ; FSK modulation activated
  '-DRF_FREQUENCY=868.950'
  '-DRTL_DEBUG=4'             ; rtl_433 verbose mode
;  '-DRTL_VERBOSE=58'          ; LaCrosse TX141-Bv2, TX141TH-Bv2, TX141-Bv3, TX141W, TX145wsdth sensor
  '-DRAW_SIGNAL_DEBUG=true'   ; display raw received messages
;  '-DMEMORY_DEBUG=true'       ; display memory usage information
;  '-DDEMOD_DEBUG=true'         ; display signal debug info
;  '-DMY_DEVICES=true'         ; subset of devices
  '-DPUBLISH_UNPARSED=true'    ; publish unparsed signal details
;  '-DRSSI_THRESHOLD=12'       ; Apply a delta of 12 to average RSSI level
;  '-DAVERAGE_RSSI=5000'       ; Display RSSI floor ( Average of 5000 samples )
;  '-DSIGNAL_RSSI=true'         ; Display during signal receive
  '-DNO_DEAF_WORKAROUND=true'
; *** RadioLib Options ***
;  '-DRADIOLIB_DEBUG=true' 
;  '-DRADIOLIB_VERBOSE=true' 
custom_description = Multi RF library with the possibility to switch between ESPilight, NewRemoteSwitch and RCSwitch, need CC1101
```

### Environment

- openmqttgateway

### Process Supervisor

not applicable

NorthernMan54, 2024-04-16:
Due to the usage of hardware based signal demodulation the sample rate configuration option is not required.  Have you enabled FSK mode ?

 [-s <sample rate>] Set sample rate (default: 250000 Hz)

eporsche, 2024-04-16:
Ok thx - good to know!  Well I set this '-DOOK_MODULATION=false' build flag and the logs showed to me that its trying to decode the signal with the  Wireless M-Bus Mode C driver.
I did try this  project https://github.com/SzczepanLeon/esphome-components and I start getting the telegrams now - not really sure what I am doing wrong

eporsche, 2024-04-18:
I need to run another test with another position in the house. I have some believe that this is/was the issue here!

closed 2024-04-18 by eporsche