NorthernMan54/rtl_433_ESP #137: Is there a way to specify only the protocols we want? Would that help with sensitivity?

melyux, 2024-04-18:
### Current Situation

RTL_433 offers a `-R` flag to specify the protocol(s) you want to decode. Is there a compile flag we can use with this library to do that too? It's not that I care about it decoding other protocols than my devices, but I'm wondering if that would help with the automatic adjustment of the signal floor (not exactly sure how that works) if it didn't have to consider all my neighbors' devices, just my own.

### Logs

```markdown
N/A
```

### Configuration

```json
N/A
```

### Environment

- Software: OpenMQTTGateway

### Process Supervisor

not applicable

### Additional Context

I love you

DigiH, 2024-04-19:
Hi @melyux 

Have you had a look at the MY_DEVICES compile definition option?

While I haven't used it myself for quite a while it was working nicely when I did, and seems to be exactly what you're looking for.

https://github.com/NorthernMan54/rtl_433_ESP?tab=readme-ov-file#compile-definition-options

NorthernMan54, 2024-04-19:
@melyux PS - Reducing the number of supported devices with the `MY_DEVICES` option would not change sensitivity, just remove potential false positives from your environment.

The decoder logic, checks all enabled decoders, and returns the results from all.  So additional decoders wouldn't stop other decoders from processing the signal.

melyux, 2024-04-19:
@DigiH @NorthernMan54 `MY_DEVICES` is just a Boolean, right? No way to specify what exact protocols I want?

But good to know that it wouldn't change anything anyone would care about in the first place

NorthernMan54, 2024-04-19:
@melyux MY_DEVICES is a compiler directive that enables you to specify which decoders to bundle and is in multiple locations within the code base.   It is best used to remove false positives from the results.  And if this isn't your issue, then it isn't worth the effort

closed 2024-11-30 by NorthernMan54