NorthernMan54/rtl_433_ESP #138: OOK_Receiver + LiLyGo LoRa32 V2.1_1.6 + Display?

grymoire, 2024-04-25:
### Current Situation

I've gotten OOK_Receiver running, but I wondered if anyone has added display output? Essentially, it would be easier to demonstrate the program if the display showed activity. 

### Proposed Change

If there is a display, then show packet activity there as well as the serial log.

### Additional Context

I might have several dongles plugged into a laptop. This would not only make it easier to me to keep track of activity, but it would also show others that is is working properly.

NorthernMan54, 2024-04-25:
The code that drives the display is part of openMQTTGateway.  Install a recent version and it will turn on

grymoire, 2024-04-25:
I did, and it displays "Connect your phone to WiFi OMG_ESP32_LORA" 
I don't want a gateway. I don't have a  MQTT server. I want a sniffer, like OOK_Receiver but I want info on the display as well as the serial port.

grymoire, 2024-04-26:
I've tried to add the display by copying piece of openMQTTGateway, but I could use some hints. I'm not a platformio user.

NorthernMan54, 2024-04-26:
The display code within OMG is actually quite complex, as it supports a lot of different message formats beyond rtl_433 and also uses internal message queuing so transplanting it from the OMG code base is very complex.  But it is based on this library - https://github.com/ThingPulse/esp8266-oled-ssd1306#f96fd6a

It should be feasible to take one of the SSD1306 examples and extend it to display the rtl_433_ESP serial messages - ie the frame 4 example https://github.com/ThingPulse/esp8266-oled-ssd1306?tab=readme-ov-file#frame-4

Or to get something going quickly, install OMG - lilygo-rtl_433 and configure wifi and MQTT.  For MQTT just install mosquito on a device that is always on.

PS Platformio is the preferred development IDE for coding ESP and other devices.   It is well worth the effort to get familiar with.

grymoire, 2024-04-26:
Okay, Thanks. I'll give it a try

closed 2024-11-30 by NorthernMan54