NorthernMan54/rtl_433_ESP #139: Cannot get any 868 MHz signals

fir3drag0n, 2024-05-07:
### Current Situation

I recently purchased a Liligo gateway and flashed it with openmqttgateway. And I set it to 868 MHz to receive my wh51 moisture sensors, but I cannot detect any signals. I flashed openmqttgateway fsk and set the frequency. Sometimes I am able to receive an acurite device, but not my moisture sensors. Is there any trick to get those signals?

I already set the frequency in openmqttgateway frontend to 868.3 MHz, but I do receive nothing. I even tried another antenna and put the wh51 device next to the Liligo gateway, but even then I don't receive the signals. I tried 3 wh51 sensors, but nothing.

Any help is appreciated!

### Logs

```markdown
No log
```

### Configuration

```json
No config
```

### Environment

- OS: openmqttgateway 
- Software: esp32

### Process Supervisor

not applicable

DigiH, 2024-05-07:
> I already set the frequency in openmqttgateway frontend to 868.3 MHz, but I do receive nothing.

Hi @fir3drag0n 

Try setting the frequency to **868.35** - are you getting anything then?

fir3drag0n, 2024-05-07:
> > I already set the frequency in openmqttgateway frontend to 868.3 MHz, but I do receive nothing.
> 
> Hi @fir3drag0n
> 
> Try setting the frequency to **868.35** MHz - are you getting anything then?

That was it. Thank you! But how do I know which frequency I have to set?

DigiH, 2024-05-08:
This is a trial and error situation and also a community effort. With Fine Offset and its re-branded Froggit, Ecowitt, Ambient Weather etc. devices the sweet spot has been found to be 868.35 MHz. 
With other devices users will currently have to tweak the frequency and test which setting will work for those.

fir3drag0n, 2024-05-08:
Is there a database or something else where I can see which frequency the devices will work on?

DigiH, 2024-05-10:
No database as of yet, but it would also require for users with all the different devices to do frequency range testing and submitting the results.

Currently the best exchange for reception experiences with OMG with rtl_433_ESPis the OMG/Theengs [community forum](https://community.openmqttgateway.com/).

Entropy512, 2024-06-28:
The original SDR-based rtl_433 will print the frequency if you use the -M level option, that's how I fed https://github.com/Entropy512/r900_esphome/blob/main/analyze_hopseq.py - although by default it only prints one decimal place.  I had to change that by modifying the source code.

It does appear to print out in higher precision if you use the CSV export without modification.

closed 2024-11-30 by NorthernMan54