NorthernMan54/rtl_433_ESP #140: LoRa32u4

coogle, 2024-05-15:
### Current Situation

I picked up a few of these boards and I am hoping I can get this project loaded up on them? Unfortunately I'm having a hard time picking the right board/platform (I think?)

https://www.amazon.com/dp/B075TWJDWG

Any pointers on getting this working? I'm currently getting this error:

`error: 'RF_MODULE_DIO2' was not declared in this scope`

### Logs

```markdown
In file included from src/rtl433-to-mqtt.h:9,
                 from src/main.cpp:26:
src/rtl433-to-mqtt.h: In member function 'virtual void Rtl433ToMqtt::setup()':
.piolibdeps/rfgw/rtl_433_ESP/src/rtl_433_ESP.h:172:37: error: 'RF_MODULE_DIO2' was not declared in this scope
 #    define RF_MODULE_RECEIVER_GPIO RF_MODULE_DIO2
                                     ^~~~~~~~~~~~~~
.piolibdeps/rfgw/rtl_433_ESP/src/rtl_433_ESP.h:172:37: note: in definition of macro 'RF_MODULE_RECEIVER_GPIO'
 #    define RF_MODULE_RECEIVER_GPIO RF_MODULE_DIO2
                                     ^~~~~~~~~~~~~~
Compiling .pioenvs/rfgw/liba09/RadioLib/modules/SX126x/STM32WLx.cpp.o
.piolibdeps/rfgw/rtl_433_ESP/src/rtl_433_ESP.h:172:37: note: suggested alternative: 'OOK_MODULATION'
 #    define RF_MODULE_RECEIVER_GPIO RF_MODULE_DIO2
                                     ^~~~~~~~~~~~~~
.piolibdeps/rfgw/rtl_433_ESP/src/rtl_433_ESP.h:172:37: note: in definition of macro 'RF_MODULE_RECEIVER_GPIO'
 #    define RF_MODULE_RECEIVER_GPIO RF_MODULE_DIO2
```

### Configuration

```json
substitutions:
  name: "rfgw"
  NAME: "RF Gateway"

packages:
  rfgw: !include rfgw.yaml

mqtt:
  broker: 10.11.12.2

wifi:
  ap:
    ssid: "rfgw-ap"
    password: "*"
  reboot_timeout: 120s
  power_save_mode: none
  fast_connect: true
  manual_ip:
    static_ip: 10.11.12.51
    gateway: 10.11.12.1
    subnet: 255.255.255.0
    dns1: 10.11.12.1
    dns2: 1.1.1.1
  networks:
    - ssid: "TSRVIOT"
      password: "*"

captive_portal:

api:
  encryption:
    key: "*"

esphome:
  name: "${name}"
  friendly_name: "${NAME}"
  libraries:
     - rtl_433_ESP=https://github.com/NorthernMan54/rtl_433_ESP.git#v0.3.0
     - RadioLib@5.7.0
  includes:
    - rtl433-to-mqtt.h
  platformio_options:
    lib_ldf_mode: "chain+"
    build_flags:
      #- '-DLOG_LEVEL=LOG_VERBOSE'
      - '-DONBOARD_LED=LED_BUILTIN'
      #- '-DPUBLISH_UNPARSED=true'
      #- '-DRAW_SIGNAL_DEBUG'
      #- '-DRTL_VERBOSE=63'
      #- '-DREGOOKFIX_DEBUG'
      #- '-DRF_MODULE_INIT_STATUS'
      - '-DRF_SX1278'
      - '-DRF_MODULE_FREQUENCY=433.92'

esp32:
  board: heltec_wifi_kit_32_V2
  variant: esp32
  framework:
    type: arduino
    version: 2.0.9
    platform_version: 6.3.0

custom_component:
  - id: rtl433
    lambda: return { new Rtl433ToMqtt("${name}") };

interval:
  - interval: 60s
    then:
      - lambda: Rtl433ToMqtt::get(rtl433)->status();
```

### Environment

- OS: macOS
- Software:
- Node:
- npm:

### Process Supervisor

not applicable

NorthernMan54, 2024-05-15:
Looking at the Amazon item, I see that this is not an Espressif ESP32 based board but an Atmega32u4 based board.  I don't think the code will work on this board, as it uses multiple cpu cores on a ESP32.  Hopefully you can still return it to amazon

But with that said, it might work if you can work out how the RA02 module is wired to the Atmega32u4 CPU and define the PINS correctly.  

_LoRa32u4 is a light and low consumption board based on the Atmega32u4 and 433MHz LoRA module RA02 from AI-Thinker and an USB battery charging circuit. Ideal for creating long range wireless networks that can exceed 2.4 GHz 802.15.4 and similar, it is more flexible than Bluetooth LE, does not require high power unlike WiFi and offers long range._

coogle, 2024-05-15:
Bah that's what I was afraid of :) Can you recommend a board available on Amazon that works? I've got a bunch of LoRa stuff laying around but none of it is what I need to give this codebase a spin. 

Side note -- I presume that there isn't any chance to getting this stuff working on an ESP8266, correct? Specifically I'm thinking the Sonoff RF bridge: https://itead.cc/product/sonoff-rf-bridge-433/

Entropy512, 2024-05-31:
FYI that recommended lilygo board is available on Amazon US, albeit at around twice the price of ordering direct from lilygo.

closed 2024-11-30 by NorthernMan54