NorthernMan54/rtl_433_ESP #145: Support selection of built-in decoders

Entropy512, 2024-06-27:
### Current Situation

Currently, this project builds in all decoders unless MY_DEVICES is set - in which case only a single decoder that is hardcoded is selected.

This leads to inefficiencies for the case where MY_DEVICES is not set, or having to manually change the hardcoded decoder when it is set

### Proposed Change

Allow users to select built-in decoders using a build-time define

NorthernMan54, 2024-06-28:
I had tried unsuccessfully to find a way to do that easily with complier directives when doing the original development work.  As the code needs a count of the number of devices to include.   Do you have some thoughts on how to achieve that ?

Entropy512, 2024-06-28:
Yeah, as I was digging through, it looks like doing it directly with compiler directives won't work, however this might:

https://docs.platformio.org/en/latest//manifests/library-json/fields/build/extrascript.html

If I'm reading that correctly, it looks like you could effectively run a variation of your rtl_433 import/update script automatically at build time to generate the DECLs in rtl_433_devices.h and signalDecoder.cpp - Probably don't want to touch those files directly, but you could probably create header fragments that are #included at the appropriate location within each file.

NorthernMan54, 2024-08-05:
If you want to submit a pull request, we can look at it.

closed 2024-11-30 by NorthernMan54