NorthernMan54/rtl_433_ESP #149: Add support for Reolink wireless doorbell

tantu07, 2024-08-14:
### Current Situation

Currently the reolink wireless doorbell when pressed is wrongly classified as Smoke-GS558. 
Product page: https://reolink.com/product/reolink-video-doorbell-wifi

Published message to MQTT:
`{
    "model": "Smoke-GS558",
    "id": 30921,
    "unit": 19,
    "learn": 0,
    "code": "4f1933",
    "protocol": "Wireless Smoke and Heat Detector GS 558",
    "rssi": -88,
    "duration": 3714864
}`

### Proposed Change

Same issue was solved for rtl_433 following https://github.com/merbanan/rtl_433/issues/2277

NorthernMan54, 2024-11-30:
I have just refreshed to rtl_433 24.10, so this should be resolved

tantu07, 2024-11-30:
@NorthernMan54  Updated to the latest version. Still issue persists.
MQTT message:
`{
    "model": "Smoke-GS558",
    "id": 30921,
    "unit": 19,
    "learn": 0,
    "code": "4f1933",
    "protocol": "Wireless Smoke and Heat Detector GS 558",
    "rssi": -55,
    "duration": 3102997
}`

Installed version:
`{
    "latest_version": "v1.8.0",
    "title": "OpenMQTTGateway",
    "release_url": "https://github.com/1technophile/OpenMQTTGateway/releases/tag/v1.8.0",
    "installed_version": "v1.8.0",
    "entity_picture": "https://github.com/1technophile/OpenMQTTGateway/raw/development/docs/img/Openmqttgateway_logo_mini_margins.png",
    "release_summary": ""
}`

Other Debug info 
`{
    "uptime": 45732,
    "version": "v1.8.0",
    "mqtt": true,
    "serial": false,
    "disc": false,
    "ohdisc": false,
    "env": "lilygo-rtl_433",
    "freemem": 107532,
    "mqttp": "1883",
    "mqtts": false,
    "mqttv": false,
    "msgprc": 1718,
    "msgblck": 0,
    "msgrcv": 1718,
    "maxq": 19,
    "cnt_index": 0,
    "minmem": 43436,
    "tempc": 52.22,
    "freestck": 3980,
    "powermode": -1,
    "eth": false,
    "rssi": -81,
    "SSID": "The Den_IoT",
    "BSSID": "D0:21:F9:EE:E6:79",
    "ip": "192.168.20.81",
    "mac": "14:2B:2F:83:49:00",
    "modules": [
        "LilyGo_SSD1306",
        "WebUI",
        "rtl_433"
    ]`

NorthernMan54, 2024-12-02:
I looked at the fix in that issue, and they are using the flex decoder which is not supported within this library.