NorthernMan54/rtl_433_ESP #150: Support for VEVOR Wireless Weather Station 7-in-1

cipriancu, 2024-08-14:
### Current Situation

With Bruno's help, we managed to decode VEVOR Wireless Weather Station 7-in-1 with rtl_433 and RLT- SDR stick
https://github.com/merbanan/rtl_433/issues/3020
but my goal is to integrate it with openmqttgateway installed on a LilyGO TTGO T3 V1.6.1 
I installed lilygo-rtl_433-fsk version but my signal is not received/decoded 

With  RLT- SDR stick  i get the data using the command: 
rtl_433 -s 1000k -Y classic -M level -f 868M -X "n=Vevor,m=FSK_PCM,s=90,l=90,r=1400,preamble=caca54aa"

what i need to do to integrate rtl_433 branch development in openmqttgateway?

### Proposed Change

add support for VEVOR Wireless Weather Station 7-in-1

toxic0berliner, 2024-10-03:
+1 looking to use this exact weather station with a Lilygo 433 so I understand there are something working settings for thé device in rtl_433, how to get those in openmqtt for thé Lilygo device ?
Any help would be greatly appreciated

hash6iron, 2024-11-04:
Hello,

I'm also interested in support for Vevor Wireless Weather Station. How can we using the LilyGo Lora32 decode FSK signal like cipriancu says above? Thanks in advance.

NorthernMan54, 2024-11-30:
I just refreshed the library to rtl_433 24.10.  Was a fix for this included?

toxic0berliner, 2024-12-02:
Seems it's not working for me, but maybe my board is broken I get no rtl_433
I've been re-building using gitpod and changing the `platformio.ini` file with thie `rtl_433_ESP = https://github.com/NorthernMan54/rtl_433_ESP.git#v0.4.0`
But no luck, I still get no RTL_433toMQTT topic at all in my mqtt broker... I do get the device pushing to mqtt for it's uptime and such... but still not very sure I managed a proper flash...

NorthernMan54, 2024-12-02:
Are you able to select the correct frequency and modulation ?

toxic0berliner, 2024-12-02:
I selected 868.300 as the frequency which seems correct from the manufacturer 's sheet, but I se me no option for modulation, only active library which is RTL_433 only other choice is inactive.
There were some settings in the platform.ini file which I didn't change, but now I think of it I think you want to make sure I uses the FSK version which I did.
But I'd still like others to test as I haven't got other RF gear and haven't been able to pickup a single RF signal over mqtt with this device so far... So maybe it's defective, maybe I built the firmware wrong...
I am also a pure noob so seeing rssi to 110 has no meaning to me, I know I tried several antenna I had lying around as well as the one sold with the lilygo device...

NorthernMan54, 2024-12-02:
Postive Results from the OMG build - https://github.com/1technophile/OpenMQTTGateway/pull/2125

DigiH, 2024-12-03:
> I selected 868.300 as the frequency which seems correct from the manufacturer 's sheet

Could you try 868.35 just for verification?

toxic0berliner, 2024-12-03:
Thanks for the help, sadly doesn't seem to change 

RFtoMQTT = {"active":3,"frequency":868.35,"rssithreshold":-106,"rssi":-113,"avgrssi":-115,"count":0,"rtl433_stack":19480,"ookthreshold":15}
I believe county says it didn't decipher a single message

DigiH, 2024-12-03:
Unfortunately the link in the decoder file only links to the US region FCC page with the 915MHz references.

I assume you do have the European/other Region 868MHz one from you above posts. 

Unfortunately with the LilyGo, the reception frequency window is not as wide as with an RLT- SDR stick, especially with 868MHZ FSK.

It took me months to find the right frequency for my own weather station, I tried 868.30, 868.40, and so forth, until someone else post about some other device from the same manufacturer stated that his works fine with the in between 868.35, and now it's working perfectly.

Does a verbose output with the RLT- SDR stick possibly give any further hints at the exact frequency?

If not, and if there is not any other issue with the vevor_7in1 for rtl_433_ESP, your best bet really is to test yourself through the frequency range in 0.05 steps.

I know it's tedious work, but if yourself, @cipriancu and @hash6iron split up the testing ranges between yourselves it shouldn't take too long to find the weed spot.

DigiH, 2024-12-03:
Also, is your weather station Crosby for this testing, or outdoors in its usual place? The LilyGos also don't have the long range reception as an RLT- SDR stick. So not that it's just a case of having to move the LilyGo closer to the station.

cipriancu, 2024-12-03:
Sorry all. I won't be able to make any test till mid December. My wheather station is working on RTL-SDR + homeassistant and the lilygo is sitting in a drawer. In terms of frequency RTL SDR sees 2:one at 868.1 and the other at 868.4. My question for whenever I'll arrive home and do the tests. Do I have to compile openmqttgateway in plaform.io to use this new library or it is included in version 1.8.0 (my lilygo is flashed with 1.7.0)

DigiH, 2024-12-03:
> 868.1 and the other at 868.4.

That's a good start! 

@toxic0berliner @hash6iron any luck with these frequencies? Or possibly anything in between in 0.05 steps?

> My question for whenever I'll arrive home and do the tests. Do I have to compile openmqttgateway in plaform.io to use this new library or it is included in version 1.8.0 (my lilygo is flashed with 1.7.0)

You will have to install a new build, but by the time you'll be back home I assume the new version of the rtl_433_ESP library has been merged into the development version of OpenMQTTGateway, so you can then just web install a development build without any need to compile anything yourself.

cipriancu, 2024-12-03:
managed to get to a computer and ive changed the conf file in HA to get the frequencies:  {"time" : "2024-12-03 21:57:43.702091", "protocol" : 263, "model" : "Vevor-7in1", "id" : 63735, "channel" : 0, "battery_ok" : 1, "temperature_C" : 4.500, "humidity" : 88, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 103, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 0, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.381, "freq2" : 868.309, "rssi" : -3.314, "snr" : 23.389, "noise" : -26.704}
{"time" : "2024-12-03 21:58:16.574055", "protocol" : 263, "model" : "Vevor-7in1", "id" : 63735, "channel" : 0, "battery_ok" : 1, "temperature_C" : 4.500, "humidity" : 88, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 96, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 0, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.381, "freq2" : 868.311, "rssi" : -3.390, "snr" : 21.431, "noise" : -24.820}
{"time" : "2024-12-03 21:58:45.157493", "protocol" : 263, "model" : "Vevor-7in1", "id" : 63735, "channel" : 0, "battery_ok" : 1, "temperature_C" : 4.500, "humidity" : 88, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 96, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 0, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.381, "freq2" : 868.298, "rssi" : -3.452, "snr" : 21.703, "noise" : -25.154}
{"time" : "2024-12-03 21:59:18.030029", "protocol" : 263, "model" : "Vevor-7in1", "id" : 63735, "channel" : 0, "battery_ok" : 1, "temperature_C" : 4.500, "humidity" : 88, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 90, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 0, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.380, "freq2" : 868.296, "rssi" : -3.490, "snr" : 20.525, "noise" : -24.015}
{"time" : "2024-12-03 21:59:46.613813", "protocol" : 263, "model" : "Vevor-7in1", "id" : 63735, "channel" : 0, "battery_ok" : 1, "temperature_C" : 4.500, "humidity" : 88, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 90, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 0, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.380, "freq2" : 868.301, "rssi" : -3.579, "snr" : 22.655, "noise" : -26.234}
{"time" : "2024-12-03 22:00:19.486996", "protocol" : 263, "model" : "Vevor-7in1", "id" : 63735, "channel" : 0, "battery_ok" : 1, "temperature_C" : 4.500, "humidity" : 88, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 90, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 0, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.381, "freq2" : 868.302, "rssi" : -3.552, "snr" : 19.784, "noise" : -23.336}
{"time" : "2024-12-03 22:00:48.070795", "protocol" : 263, "model" : "Vevor-7in1", "id" : 63735, "channel" : 0, "battery_ok" : 1, "temperature_C" : 4.400, "humidity" : 88, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 90, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 0, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.381, "freq2" : 868.294, "rssi" : -3.555, "snr" : 19.558, "noise" : -23.113}
{"time" : "2024-12-03 22:01:20.944513", "protocol" : 263, "model" : "Vevor-7in1", "id" : 63735, "channel" : 0, "battery_ok" : 1, "temperature_C" : 4.500, "humidity" : 88, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 90, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 0, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.381, "freq2" : 868.302, "rssi" : -3.567, "snr" : 23.015, "noise" : -26.581}
{"time" : "2024-12-03 22:01:49.528426", "protocol" : 263, "model" : "Vevor-7in1", "id" : 63735, "channel" : 0, "battery_ok" : 1, "temperature_C" : 4.400, "humidity" : 88, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 90, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 0, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.381, "freq2" : 868.292, "rssi" : -3.433, "snr" : 17.956, "noise" : -21.389}
{"time" : "2024-12-03 22:02:22.400706", "protocol" : 263, "model" : "Vevor-7in1", "id" : 63735, "channel" : 0, "battery_ok" : 1, "temperature_C" : 4.400, "humidity" : 88, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 90, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 0, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.381, "freq2" : 868.310, "rssi" : -3.572, "snr" : 26.019, "noise" : -29.591}
{"time" : "2024-12-03 22:02:50.984290", "protocol" : 263, "model" : "Vevor-7in1", "id" : 63735, "channel" : 0, "battery_ok" : 1, "temperature_C" : 4.400, "humidity" : 88, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 90, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 0, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.381, "freq2" : 868.286, "rssi" : -3.428, "snr" : 23.945, "noise" : -27.373}
{"time" : "2024-12-03 22:03:23.856898", "protocol" : 263, "model" : "Vevor-7in1", "id" : 63735, "channel" : 0, "battery_ok" : 1, "temperature_C" : 4.400, "humidity" : 88, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 90, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 0, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.381, "freq2" : 868.305, "rssi" : -3.672, "snr" : 17.904, "noise" : -21.575}
{"time" : "2024-12-03 22:03:52.440607", "protocol" : 263, "model" : "Vevor-7in1", "id" : 63735, "channel" : 0, "battery_ok" : 1, "temperature_C" : 4.400, "humidity" : 88, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 90, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 0, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.381, "freq2" : 868.309, "rssi" : -3.581, "snr" : 21.842, "noise" : -25.423}
[... cut ...]
{"time" : "2024-12-03 22:06:49.075418", "enabled" : 1, "since" : "2024-12-03T21:56:49", "frames" : {"count" : 3764, "fsk" : 17, "events" : 17}, "stats" : [{"device" : 263, "name" : "Vevor Wireless Weather Station 7-in-1", "events" : 17, "ok" : 17, "messages" : 17}]}
{"time" : "2024-12-03 22:06:56.809287", "protocol" : 263, "model" : "Vevor-7in1", "id" : 63735, "channel" : 0, "battery_ok" : 1, "temperature_C" : 4.400, "humidity" : 88, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 80, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 0, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.381, "freq2" : 868.298, "rssi" : -3.639, "snr" : 17.974, "noise" : -21.613}

NorthernMan54, 2024-12-03:
Humm, the difference between Freq1 and Freq2 is approx 0.080 Mhz

Am wondering if changing this to 80, this may make a difference if it doesn't work once the frequency is dialled in - https://github.com/NorthernMan54/rtl_433_ESP/blob/91bd22a80d398de49fcc0af0375d187cdfb455dc/src/rtl_433_ESP.cpp#L226

toxic0berliner, 2024-12-04:
I tried from 868.1 to 868.45 in .05 increments, nothing yet.
Since I'm a bit worried I built something wrong, I would like to attempt to re-flash the openMQTT dev using the web tools but I don't see any option for dev on the page https://docs.openmqttgateway.com/upload/web-install.html#using-openmqttgateway
I also tried 915.000 on the offchance I had an US version shipped to me in France by mistake, doesn't seem to be the case.

Sorry for the noob-level help needed. I'm considering adding my own option in the html dropdown if I can findout the path for the dev builds 🤣 I'm fairly goot at web, shell, but real new to ESP32 & especially RF 😉 

Meanwhile I have put the lilygo device on a battery pack and it's sitting outside just next to the weatherstation, can't be any closer. I am going over the same range with .005 increments now and fiving it more time inbetween.

Maybe if you have one piece of advice : I plugged in a big antenna I bought a while ago that was for WIFI I expect 2.4GhZ on the liligo device thnking bigger is better, would you advise I switch back to the little antenna provided with the lilygo instead ?

toxic0berliner, 2024-12-04:
silly me, just found the url in the github actions : https://docs.openmqttgateway.com/dev/
installing now...

DigiH, 2024-12-04:
Northernman is likely asleep in his part of the world right now, so I've done the following
• Created a branch with the 0.4.0 release and added his suggested change above
• Started a OpenMQTTGateway development build with that branch, which should be ready in about 90 minutes from now

You can web upload the development test build once it changes to **SHA:81eb8b** at the top of the dev upload page.

Let's see what it brings.

toxic0berliner, 2024-12-05:
oups, sorry, I didn't anticipate this last message to be time-sensitive and for sure now I look at it it's alredy gone, sorry for that and thanks for the efforts DigiH
I will be watching this topic more closely and have the device on my desk in case you pull the same strings again, else I will be patiently waiting for the [MR](https://github.com/1technophile/OpenMQTTGateway/pull/2125) to be merged, after all it's more my fault I'm not comfortable building&flashing ESP devices 😁

DigiH, 2024-12-05:
It wasn't so much about 0.4.0 being merged, as you seemed to have done that already fine with your gitpod firmware build, but more to also include the above change from @NorthernMan54.

I have started the test build again, and it will be available with the same SHA in about 90 minutes, again only until just after midnight ;)

toxic0berliner, 2024-12-05:
git it !
flashed and configures, it's running, so far not much more luck but I'll try to monitor and maybe sweep across the frequency range again:
```
{"active":3,"frequency":868.381,"rssithreshold":-106,"rssi":-114,"avgrssi":-115,"count":0,"rtl433_stack":19480,"ookthreshold":15}
```

cipriancu, 2024-12-09:
Hi all 
got home and started to work firstly i checked again the frequency with HDSDR 
[image]
and the 2 frequencies are 868.307 and 868.376 

next, i flashed [SHA:1b50c8 TEST ONLY] with lilygo-rtl_433-fsk, i set different frequencies around both values 868.307, 868.31, 868.376, 868.38, 868.37,868.35  ( in this time I had the RTL-SDR turned on to check if I receive something)  and on lilygo no success

DigiH, 2024-12-09:
Hin @cipriancu 

> next, i flashed [SHA:1b50c8 TEST ONLY]

That was the nightly development build again, with rtl_433_ESP 0.4.0, but without
state = radio.setFrequencyDeviation(40)
as from @NorthernMan54's suggestion above.

Let me start that build again, and in about 90 minutes you should be able to install the **SHA:81eb8b** build.

cipriancu, 2024-12-09:
in the meantime i flashed a via Platform IO https://github.com/1technophile/OpenMQTTGateway/tree/rtl_433_esp-v0.4.0 
i set the frequency at 868.35 and it seems to work , 

N: [ OMG->MQTT ] topic: home/OMG_lilygo_rtl_433_ESP_FSK/RTL_433toMQTT/Vevor-7in1/0/63735 msg: {"model":"Vevor-7in1","id":63735,"channel":0,"battery_ok":1,"temperature_C":24.8,"humidity":49,"wind_avg_km_h":0,"wind_max_km_h":0,"wind_dir_deg":185,"rain_mm":0.466,"uv":0,"light_lux":11,"mic":"CHECKSUM","protocol":"Vevor Wireless Weather Station 7-in-1","rssi":-33,"duration":83996} 

but it doesn't seems to capture all packets at every 20 seconds, i need to test a bit more

DigiH, 2024-12-09:
> but it doesn't seems to capture all packets at every 20 seconds, i need to test a bit more

See how it goes when the **SHA:81eb8b** build is ready, with the wider frequency deviation.

toxic0berliner, 2024-12-09:
I have 81eb8b I have been swiping the frequencies, so far no luck trying again around 868.35

DigiH, 2024-12-09:
> I have 81eb8b I have been swiping the frequencies, so far no luck trying again around 868.35

I meant your actual LilyGo board, which version is the board hardware?

toxic0berliner, 2024-12-09:
Wow, something new: 
`{"active":3,"frequency":868.35,"rssithreshold":-106,"rssi":-112,"avgrssi":-115,"count":1819,"rtl433_stack":5048,"ookthreshold":15}`
The count is not 0 😯
Still, I don't see any new device in my mqtt integration it seems and the console doesn't log anything having submitted new info to mqtt between the count=0 and this 1819.
It seems stuck a 1819 now...
Maybe you've found my issue though:
[image]

toxic0berliner, 2024-12-09:
I believed my device to be 433 back then.
Maybe you could confirm I would have better luck with this one: https://lilygo.cc/products/lora3?variant=42476923682997

DigiH, 2024-12-09:
I'm afraid you'll need a **868/915MHZ** version, where yours now says **433/470MHz**.
That's why you are not seeing any reception of your 868MHz station.

https://de.aliexpress.com/item/1005003062523617.html

DigiH, 2024-12-09:
Cross over ;) yes, you would be a lot luckier with a 868MHz LilyGo 😃

DigiH, 2024-12-09:
@cipriancu - the **SHA:81eb8b** build is ready for a test drive …

DigiH, 2024-12-09:
Didn't you mention you used GitPod before?

I basically have forked this repo here, with the main branch being the 0.4.0 state right now. Once forked just create a new branch and in that branch make the change @NorthernMan54 suggested above - maybe best wait to see if @cipriancu comes back with good results first ;)

Then in OpenMQTTGateway, either forked as well or though GitPod again, all you have to do is redirect the rtl_433 library link to your own branch, as I did in the test build with
```
; rtl_433_ESP = https://github.com/NorthernMan54/rtl_433_ESP.git#v0.3.3
rtl_433_ESP = https://github.com/DigiH/rtl_433_ESP.git#freqdev
```
in Platform.ini. Then build and upload - 0.4.0 with the above frequency deviation change.

Alternatively just remove the branch tag `#v0.3.3` at the end of the existing URL or change it to `#0.4.0` to get the current state as is.

I hope it makes sense.

cipriancu, 2024-12-09:
ALL
firstly related [v0.4.0] 
between 12:58 and 14:34, I captured 53 packets from 288 
i installed now 
[SHA:81eb8b TEST ONLY](https://docs.openmqttgateway.com/dev/)

but I did not receive anything 
@DigiH are you sure you've committed the changes for the frequency deviation?
here https://github.com/DigiH/rtl_433_ESP  i see that the last commit was 5 days ago 

with both [v0.4.0] or SHA:81eb8b home assistant is give me this 

[image]

I'm thinking of downloading OpenMQTTgateway v1.8.0 and changing in the platform.io the link to rtl_433_ESP either to DigiH or NorthernMan54

DigiH, 2024-12-09:
> @DigiH are you sure you've committed the changes for the frequency deviation?
> here https://github.com/DigiH/rtl_433_ESP i see that the last commit was 5 days ago

If you look at the commit from 5 days ago, I only added what @NorthernMan54 suggested above 5 days ago

https://github.com/NorthernMan54/rtl_433_ESP/compare/main...DigiH:rtl_433_ESP:freqdev

but at the same time also wondering if the second occurrence of radio.setFrequencyDeviation(40) at

https://github.com/DigiH/rtl_433_ESP/blob/freqdev/src/rtl_433_ESP.cpp#L276

should also be changed. As only one was mentioned I only changed the one. I'm not very versed with the radio tuning intricacies and honestly wasn't too sure about exactly what I was changing, but I just wanted to make the above suggested change available for testing ;)

So you can see that the change is included, and if it makes it worse and not better for you I suppose some other tweaking might be required to get a better reception than your current 53 out of 288.

NorthernMan54, 2024-12-09:
@DigiH A bit of a moment for me there, if you look at the compiler directives, the first radio.setFrequencyDeviation is for the CC1101 based build, and the second for the SX127X based build.  So it should be the second one

DigiH, 2024-12-09:
@NorthernMan54 You know me, I do my BLE decoding and am used to creating test builds for that, but when it comes to the rtl_433 decoding I just went to the line you stated above ;P without actually look too closely as I was also working on more BLE decoders at the same time.

So @cipriancu better change the second one then, possibly keeping the 1st for CC1101 requiring the same or similar change, if any of the Vevor7in1 owners do have a 868MHz CC1101?!?

Where would all the fun and excitement be without small hiccups like that along the way.

DigiH, 2024-12-09:
> or do you quickly want me to make the change in my branch, so you can just link to it from Platform.ini in your OMG build?

Done, both now with 80

https://github.com/NorthernMan54/rtl_433_ESP/compare/main...DigiH:rtl_433_ESP:freqdev

so yo9u can just keep the rtl_433 link from the OMG test branch

`rtl_433_ESP = https://github.com/DigiH/rtl_433_ESP.git#freqdev`

cipriancu, 2024-12-09:
thanks, DigiH
I compiled Platform.ini the old version ( i think it was with 40) and I got 13 records in 15 minutes 
then I compiled this new version and I got 20 records in 40 minutes,

but I wonder if there isn't a processing of the data that dont push a record in MQTT as long nothing is changing. because i have the weather station next to me inside the changes are minimal

DigiH, 2024-12-09:
> but I wonder if there isn't a processing of the data that dont push a record in MQTT as long nothing is changing. because i have the weather station next to me inside the changes are minimal

AFAIK that should not happen that identical value messages are being filtered. I think you said that your station broadcasts every 20 seconds, so you really should ideally see a received and decoded message envy 20 seconds, even if and when the values do not change. At least that is what I was seeing with my weather station during initial testing and setup. So unless something has changed in regards to that it should still behave the same.

Could it be that the 'missing' decodings might actually fall into another decoder. I have this with my fridge/freezer thermometers, that every now and then they get recognised and decoded as some other brand thermometer, but I do think that is actually unique to my thermometers, as there was quite a long issue thread about these particular ones - just a thought though if you have other devices detected, if one of them might look very similar to your station.

13 records in 15 minutes - with braodcsdsts every 20 seconds, correct? so 13 recovers from 45 broadcast!?! 29% hits

20 records in 40 minutes - 20 out of 120 broadcasts. Only 17%

So it did actually get worse with the change, if the first was with 0.4.0 as is in the PR, and the seconds with 0.4.0 plus the 
radio.setFrequencyDeviation change to 80. You did only compile after I submitted the change at the actually required second location, right?

As I said, I'm happy to help with creating forks and running test builds, but with the actual rtl_433 RF decoding side it's up to your guys to come up with what else might be worth trying ;) but I also have to say that the first 29, let's round it to 30% I don't find too bad actually, if correct receptions come in on an average of every 60 seconds, an interval some other stations actually broadcast as their default.

DigiH, 2024-12-09:
@cipriancu 

Now with the frequency deviation set higher, what base frequency setting did you actually use? Still 868.35?

Looking any your list of sample further above, it looks as if the base frequnecy1 is always very close around **868.40**

Maybe try another round with setting the frequency in the webUI to that??

cipriancu, 2024-12-10:
Thanks, DigiH, for the detailed analysis.

I don’t think the frequency deviation needs to be set to 80. I believe it should stay at 40 because the two frequencies emitted by the weather station are 868.307 and 868.376.

I changed the base frequency to 868.34 since that, with a deviation of ±40, should cover the full spectrum of the weather station’s signals. With ±80, I suspect it might capture more noise, which could be affecting the decoding rate.

Let me know what you think, and I’ll continue testing with this setup.

[image]

DigiH, 2024-12-10:
I honestly don't know enough about the RF rtl_433 decoding intricacies to know what other parameters might be helpful to try out and tweak, to possibly improve on your 30% reception and decoding rate which you had with the original frequency deviation 40 setting.

I'm happy to help though with creating test build with any such changes.

toxic0berliner, 2024-12-20:
I just recieved the Lilygo 868 !!!
Built the latest lilygo-rtl_433-fsk thing available with gitpod by replacing as below : 
```
# rtl_433_ESP = https://github.com/NorthernMan54/rtl_433_ESP.git#v0.4.0
rtl_433_ESP = https://github.com/DigiH/rtl_433_ESP.git#freqdev
```
Recieved a msg from the weather station right away, seems they are indeed not coming very frequently but will be monitoring over some time.
Seems to me also there's at least an issue in humidity % that is over 100% and another in wind direction that should be 0<x<360 or -180<x<180
[image]
Will look into it when I got some time.

Thanks for the efforts all of you !

NorthernMan54, 2024-12-20:
The CRC logic in the decoder is here - https://github.com/NorthernMan54/rtl_433_ESP/blob/64f05666a96b2204f347c3ef14e664f5b912cd12/src/rtl_433/devices/vevor_7in1.c#L90

And looking at the logic it is already discarding them.

I wonder if the same data quality issue exists with using rtl_433 and a sdr ?

NorthernMan54, 2024-12-22:
Also this compiler directive may give more insights into the signal decoding

```
RTL_VERBOSE=96
```

PS - I haven't used this option since the most recent update, fingers crossed it still works with the update

toxic0berliner, 2024-12-22:
To be honest I'm going to be lazy now, I find that the base station is not sendig over RF, and it is doing quite a lot for me in fact:
- it has the pressure sensor
- it averages the values or does some magic to avoid the few but unmistakable erroneous values (300k lux, 1800°C, ...)
- it gives a rain rate (mm/h) over the last minute while the RF only gets the total mm since startup
I could fix all of that with a bit of work in home assistant, but hey, the base station already does it for me and I already put the work into making the man-in-the-middle attack on the base station to keep it local instead of pushing to the cloud, so...
So I fear I'm gonna be lazy but for what it's worth I do believe it's all working properly now.
Maybe one could enlighten me as to if -70 is a good RSSI or not, I could tryout another antenna or move the lilygo closer, beyond that I fear I'm not going to rebuild. 
But this should be confirmation enough to actually merge and get the new release someday live for everyone else

DigiH, 2025-01-01:
Any chance to get a raw MQTT message at all from you guys? 

Just wondering if there is some information for which we might still be missing some device class definitions for proper discovery, as it looks like there might also be a solar radiation value.

RealGue, 2025-08-04:
I also have problems receiving messages from my Vevor weather station.

I'm using a Lilygo V2.1 with the lilygo-rtl_433-FSK image of OpenMQTTGateway (from the website).

I receive packets using a RTL2832 and rtl_433 directly:

{"time" : "2025-08-04 15:51:25", "protocol" : 263, "model" : "Vevor-7in1", "id" : 29739, "channel" : 0, "battery_ok" : 1, "temperature_C" : 19.500, "humidity" : 73, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "wind_dir_deg" : 108, "rain_mm" : 0.000, "uv" : 0, "light_lux" : 1088, "mic" : "CHECKSUM", "mod" : "FSK", "freq1" : 868.372, "freq2" : 868.109, "rssi" : -0.134, "snr" : 16.400, "noise" : -16.533}

But when using OpenMQTTGateway I don't receive anything. I tried frequencies from 867.500 to 869.000 in 0.05 steps. And confirmed each time that a packet was sent through rtl_433.
I can see other weather stations.

Any ideas?

DigiH, 2025-08-04:
> I'm using a Lilygo V2.1 with the lilygo-rtl_433-FSK image of OpenMQTTGateway (from the website).

@RealGue  Which LilyGo 2.1 version do you have? Can you post a photo of the back, with the chip visible?

In general the LilyGos don't have as long a reception range as any RTL-SDR stick, so the distance might also be an issue.

RealGue, 2025-08-04:
@DigiH Actually I used the Web uploader which used the https://ota.openmqttgateway.com/binaries/v1.8.1/lilygo-rtl_433-fsk-firmware.bin binary - which is probably not the development version.
To use the dev version I have to install Gitpod as described in https://docs.openmqttgateway.com/upload/gitpod.html, correct?

I've added a photo of my LilyGo.

[image]

luissiopa, 2025-12-24:
@DigiH I'm trying to receive packets but no luck, probably I'm using wrong version. I have the TTGO T3_V1.6.1 like the picture from @RealGue. 
With RLT- SDR stick and the rtl_433 library I can receive packets, but with the TTGO nothing.

What I'm doing wrong? Can you help me to get the right version? Can the upload be also via Web uploader?

closed 2025-12-25 by NorthernMan54