NorthernMan54/rtl_433_ESP #151: Several rtl_433 sensors are sporadically broadcasting a spike of nonsensical temperature or battery data after upgrade

puterboy, 2024-08-19:
### Analysis

I reported this bug on OpenMQTT Gateway since I am using their direct code but my understanding is that the rtl_433_esp stuff is based on your code so if it's ok, I would like to report the bug here too since you are likely more knowledgeable about the changes.

I have a bunch of cheap LaCrosse, AmbientWeather, and Nexus temperature & humidity sensors that I read using a LILYGO LORA 433 esp32 running OpenMQTT Gateway compiled under the lilygo-rtl_433 environment.

Everything was working properly until I upgraded to v175 of OpenMQTT Gateway.
I started noticing that multiple of the sensors would sporadically (without any seeming pattern) transmit erroneous MQTT data sensor data.

For example, my 4 older LaCrosses sensors randomly broadcast a temperature of 2 degC (33.8 degF) regardless of the actual room temperature (note the erroneous value is always exactly 2 deg C). The newer ones are a different model and seem to be OK

Similarly, the battery level would go from a normal 100.0 (corresponding to 100%) to a non-sensical but value that seems different for each sensor (vs. temperature where 33.8 was always the error value). Examples include: 21385.0, 20989.0, 12079.0, 21682.0, 8218.0, 2575.0

In both cases, the erroneous spikes last for only a single reading before returning to normal and then potentially spiking again in an hour or two.

I can't see any pattern in the timing of the spikes or even which sensors are spiking (some of them seem to not be spiking or it could just be that I haven't observed them long enough since the update)

Reverting to the prior version (development branch with last commit on 3/5/23) removes this issue -- so it very much seems to be a problem with the latest code and not with my setup or sensors.

### Expected Behavior

MQTT messages should represent the same temperature/humidity/rssi/battery data as transmitted by the various sensors at 433 MHz.

### Steps To Reproduce

As above

### Logs

```markdown
I don't see any abnormal messages
```

### Configuration

```json
As above
```

### Environment

NA

### Process Supervisor

not applicable

puterboy, 2024-08-25:
Interestingly, the version that works (dev branch from March 2024) interestingly uses a newer version of the library rtl_433_ESP  (rtl_433_ESP.git#v0.3.2 than the current August dev version (rtl_433_ESP.git#v0.3.1) that creates the spurious spikes

Any idea why the library version has been downgraded?

The only seemingly significant differences (beyond spelling corrections in comments) are as follows:
```
--- OpenMQTTGateway-old/.pio/libdeps/lilygo-rtl_433-jjk/rtl_433_ESP/src/rtl_433/r_api.c 2024-03-13 23:51:45.000000000 -0400
+++ OpenMQTTGateway-release/.pio/libdeps/lilygo-rtl_433-jjk-ota/rtl_433_ESP/src/rtl_433/r_api.c 2024-08-18 18:35:00.104503500 -0400
@@ -798,12 +798,10 @@
       else if ((d->type == DATA_DOUBLE) &&
                (str_endswith(d->key, "_in") || str_endswith(d->key, "_inch"))) {
         d->value.v_dbl = inch2mm(d->value.v_dbl);
-        // need to free ptr returned from str_replace
-        char* new_label1 = str_replace(d->key, "_inch", "_in");
-        char* new_label2 = str_replace(new_label1, "_in", "_mm");
-        free(new_label1);
+        char* new_label =
+            str_replace(str_replace(d->key, "_inch", "_in"), "_in", "_mm");
         free(d->key);
-        d->key = new_label2;
+        d->key = new_label;
         char* new_format_label = str_replace(d->format, "in", "mm");
         free(d->format);
         d->format = new_format_label;
```
And
```
--- OpenMQTTGateway-old/.pio/libdeps/lilygo-rtl_433-jjk/rtl_433_ESP/src/rtl_433_ESP.cpp 2024-03-13 23:51:45.000000000 -0400
+++ OpenMQTTGateway-release/.pio/libdeps/lilygo-rtl_433-jjk-ota/rtl_433_ESP/src/rtl_433_ESP.cpp 2024-08-18 18:35:00.104503500 -0400
@@ -32,12 +32,8 @@
 #if defined(RF_MODULE_SCK) && defined(RF_MODULE_MISO) && \
     defined(RF_MODULE_MOSI) && defined(RF_MODULE_CS)
 #  include <SPI.h>
-#  if CONFIG_IDF_TARGET_ESP32C3 || CONFIG_IDF_TARGET_ESP32S3
-SPIClass newSPI(FSPI);
-#  else
 SPIClass newSPI(VSPI);
 #  endif
-#endif

 #ifdef RF_SX1276
 SX1276 radio = RADIO_LIB_MODULE;
@@ -59,14 +55,14 @@

 /*----------------------------- rtl_433_ESP Internals -----------------------------*/

-#define rtl_433_ReceiverTask_Stack    2048
+#define rtl_433_ReceiverTask_Stack    2000
 #define rtl_433_ReceiverTask_Priority 2
 #define rtl_433_ReceiverTask_Core     0

-/*----------------------------- Initialize variables -----------------------------*/
+/*----------------------------- Initalize variables -----------------------------*/

 /**
- * Is the receiver currently receiving a signal
+ * Is the receiver currently receving a signal
  */
 static bool receiveMode = false;

```
Could the change in `ReceiverTask_Stack` be causing the problem due perhaps to overflow???
After all I do have a couple of dozen sensors...

NorthernMan54, 2024-08-25:
Try running just the example code from here on one of your devices.   It likely does not show an issue.

puterboy, 2024-08-25:
You mean flash this repo to my LORA device rather than 1technophile's version?

Assuming it works here, any idea where the problem may be in 1technophile's version so I can try to figure it out?

NorthernMan54, 2024-08-25:
Yes, open VSCODE in this directory - https://github.com/NorthernMan54/rtl_433_ESP/tree/main/example/OOK_Receiver

And edit platformio to match your device.

puterboy, 2024-08-26:
The problem seems to be a concurrency issue relating to how messages are queued in OpenMQTT gateway -- and not related to the rtl_433.
As such, I will close this thread..

closed 2024-08-26 by puterboy