NorthernMan54/rtl_433_ESP #153: Is Fine Offset WS69 supported?

mcmlxxvii, 2024-09-24:
### Current Situation

I see that there's a registering protocol [24] already in place for "Fine Offset Electronics WS80 weather station", so I'm wondering if an [Ecowitt WS69](https://shop.ecowitt.com/en-ca/products/ws69) is also supported by such protocol, as they both use FSK modulation and have similar "features" (naive question).

I have a 915Mhz Heltec V2 running OpenMQTTGateway working exceptionally well with a couple WH51, thanks to the rtl_433_ESP library. Therefore a WS69 would be a great addition to my setup, if supported.

If not, are there any plans to support it in a near future?

Thanks!

### Logs

```markdown
N/A
```

### Configuration

```json
N/A
```

### Environment

N/A

### Process Supervisor

not applicable

NorthernMan54, 2024-09-24:
Support for devices actually comes from the rtl_433 code base, so once they support it, then I should be able to update my library to support it as well. I did a quick scan this AM, and did not see that support was listed within rtl_433, but I could be wrong.

mcmlxxvii, 2024-09-24:
Thank you for the quick reply here as well! :)

I'll ask about upstream support in rtl_433's GIT. 👍

closed 2024-09-24 by mcmlxxvii

fir3drag0n, 2024-12-30:
Any news on that?