NorthernMan54/rtl_433_ESP #160: Sync w/ upstream rtl_433 24.10 or higher?

claudobahn, 2024-11-24:
### Current Situation

According to the README.md, this library is currently based on 22.11. In particular, it's missing support for newer Fine Offset sensors such as the WS90 (see issue #111 ).

### Proposed Change

Work through `tools/generate_include_src.sh` (in particular util.h has been split up into multiple smaller util includes) before running `tools/update_rtl_433_devices.sh`.

I took a stab at it, but I'm unsure what to do about `These src files need copying and updating` and `These include files need copying and updating`.

### Additional Context

@NorthernMan54 let me know how I can help...

As mentioned, at a minimum util.c and util.h have to be removed and at a minimum c_util.h has to be pulled in, though other parts of it, too, probably.

The output of `tools/generate_include_src.sh`:

```
./generate_include_src.sh 
Only run after making major changes to reset list of include and source files
Include Files to check

abuf.h
bitbuffer.h
compat_time.h
data.h
decoder.h
decoder_util.h
fatal.h
list.h
logger.h
optparse.h
output_log.h
pulse_analyzer.h
pulse_data.h
pulse_detect.h
pulse_slicer.h
r_api.h
r_device.h
r_private.h
r_util.h
rfraw.h
rtl_433.h
util.h

File abuf.h was not changed

File bitbuffer.h was changed

File compat_time.h was changed

File data.h was changed

File decoder.h was changed

File decoder_util.h was changed

File fatal.h was changed

File list.h was not changed

File logger.h was not changed

File optparse.h was not changed

File output_log.h was changed

File pulse_analyzer.h was changed

File pulse_data.h was changed

File pulse_detect.h was not changed

File pulse_slicer.h was not changed

File r_api.h was changed

File r_device.h was changed

File r_private.h was changed

File r_util.h was not changed

File rfraw.h was not changed

File rtl_433.h was changed

diff: ../rtl_433/include/util.h: No such file or directory
File util.h was changed

-----------------------------------------------------------------------------------------

Source Files to check

abuf.c
bitbuffer.c
compat_time.c
data.c
decoder_util.c
list.c
logger.c
output_log.c
pulse_analyzer.c
pulse_data.c
pulse_slicer.c
r_api.c
r_util.c
util.c

Checking  abuf.c
File abuf.c was changed

Checking  bitbuffer.c
File bitbuffer.c was changed

Checking  compat_time.c
File compat_time.c was changed

Checking  data.c
File data.c was changed

Checking  decoder_util.c
File decoder_util.c was changed

Checking  list.c
File list.c was changed

Checking  logger.c
File logger.c was not changed

Checking  output_log.c
File output_log.c was changed

Checking  pulse_analyzer.c
File pulse_analyzer.c was changed

Checking  pulse_data.c
File pulse_data.c was changed

Checking  pulse_slicer.c
File pulse_slicer.c was changed

Checking  r_api.c
File r_api.c was changed

Checking  r_util.c
File r_util.c was changed

Checking  util.c
diff: ../rtl_433/src/util.c: No such file or directory
File util.c was changed
```

NorthernMan54, 2024-11-26:
I have started the refresh a few times, and keep getting stuck.  The scripts were written at a particular point in time against the rtl_433 codebase, and as things change over time they break.  So majority of the update now needs to be done by hand.  I had envisioned that the code base would only have minor tweaks over time and not significant refactoring 

when I built the library I kinda split the rtl_433 codebase, between the raw signal receive and demodulation logic and signal decode logic.  As the boards we are using, provide a demodulated signal, we only need access to the decode logic.

maybe for long term support, we should have our own code to manage the decoding, and just leverage the device decoders. Or take a different approach for the refresh.

NorthernMan54, 2024-11-30:
I went back and looked at it again, and took a different approach this time, and only pulled the device decoders over, and left the rest behind.  Besides a couple of minor missing functions and definitions it actually worked.  So I just published it as 0.4.0.

closed 2025-09-05 by NorthernMan54