NorthernMan54/rtl_433_ESP #165: Need help to read water meter R900: Build and deploy the library

uriva9w, 2025-01-07:
### Current Situation

I got a LILYGO® TTGO Meshtastic T-Beam V1.2 ESP32 LoRa 915MHz, I want to deploy your library to read my water meter R900 (900 MHz).  Follow the "**Building and implementing the sample**". I selected the lilygo- Lora32.  I have some questions
Do I have to set some Compiling paremeters for the sensor I want to read?
Once built..ho do I use (load) the library in Arduino IDE?

### Logs

```markdown
Nothing to log
```

### Configuration

```json
LILYGO® TTGO Meshtastic T-Beam V1.2 ESP32 LoRa 915MHz with a water meter R900
```

### Environment

- OS:
- Software: W11 Pro
- Node:
- npm:

### Process Supervisor

not applicable

NorthernMan54, 2025-01-08:
Pls use either an openmqttgateway build for your setup, or use the OOK receiver code in the examples directory.

for building pls use vscode and platformio

uriva9w, 2025-01-09:
Thx for your reply
1- I used VS with Platformio and I built..I got 
[image]
2- I'll use the OOK receiver code in the examples directory
My question , how to load the libraries ArduinoJson.h, ArduinoLog.h and rtl_433_ESP.h in Arduino IDE?

NorthernMan54, 2025-01-09:
Pls use vscode and platformio, that will load the correct libraries as needed.

and as I don’t use the arduino ide i cant help you with that

uriva9w, 2025-04-12:
Hi... I wanted to know if somebody had success reading R900 water meter.

jmiller0, 2025-10-02:
Also came here looking for the same, currently using a RTL-SDR usb dongle to read R900 and was hoping for a bridge with mqtt support.  From my research it doesn't exist or not possible due to high bitrate signals.