NorthernMan54/rtl_433_ESP #167: OOK and FSK should be both supported at the same time

serialrf433, 2025-01-22:
### Current Situation

Currently you are forced to choose if OOK or FSK is enabled. This is not what most users coming from rtl_433 expect. I was surprised why my reception (FSK transmitter) was not working. I had to spend much time to understand that for some unknown reason there is a switch between OOK and FSK rtl_433_esp .
In platformio.ini there is:
  '-DOOK_MODULATION=true'      ; False is FSK, True is OOK

### Proposed Change

Please remove the need of  
 '-DOOK_MODULATION=true'      ; False is FSK, True is OOK

OOK and FSK should be both just working at same time like known from the rtl_433 project users.

NorthernMan54, 2025-01-22:
The requirement to select FSK or OOK modulation is needed to tell the radio chipset which mode to operate in.

serialrf433, 2025-01-26:
I think my request was a step too far.
I understand that selection of FSK or OOK is required for a typical radio chipset. I was using SDR most time and that is why i was thinking both reception types would be possible at the same time.

Its not that simple on bigger projects using this rtl_433_esp libraty to recompile the whole project with this library when a switch from OOK to FSK is required.
Could you still remove the need of  '-DOOK_MODULATION=true' and solve that way the need of recompilation? The library should be able to use FSK and OOK when compiled by default. Some option to switch between both would be great.

Most devices today use FSK. But OOK is default on this library. I expect many people would want to use FSK instead of OOK.

juanboro, 2025-06-14:
fyi - I created a fork library based on the excellent work done here that does JUST the rtl_433 decoding [here](https://github.com/juanboro/rtl_433_Decoder_ESP) - to eliminate the interdependencies on the radio and mqtt clode.  It allows for dynamically enabling/disabling decoders during runtime (by default both are enabled).  Although - I've personally had no success getting my own CC1101 to reliably receive FSK yet.  But maybe it can handle your specific needs.

closed 2026-05-11 by NorthernMan54