NorthernMan54/rtl_433_ESP #168: Question: Would SX126x support be possible on FSK mode?

serialrf433, 2025-01-26:
### Current Situation

I have read the reason for not supporting the SX126x here: https://github.com/NorthernMan54/rtl_433_ESP/issues/82#issuecomment-1528044564 
and here
https://github.com/NorthernMan54/rtl_433_ESP/issues/46#issuecomment-1325960788

But when using FSK and not OOK, would it then be possible to use the chip with this library? Those both options are also available in the datasheet of the SX126x:
https://github.com/NorthernMan54/rtl_433_ESP/blob/main/src/rtl_433_ESP.cpp#L959

### Logs

```markdown
None
```

### Configuration

```json
None
```

### Environment

None

### Process Supervisor

not applicable

ldina, 2025-02-28:
interested too, I have one device with that damned chip that seems compatible to nothing

ldina, 2025-03-14:
@serialrf433 thanks a lot for taking the time and educating me about the chip history. I really like your product.

NorthernMan54, 2025-03-14:
@serialrf433 under the covers the rtl_433 device decoders are using the raw bit stream from the radio receiver chipset.  To use packet mode instead, this library and the rtl_433 device decoders would need to be rewritten.  So likely feasible, but it would be a different code base, and likely coded to a particular sensor’s FSK encoding scheme.

closed 2025-09-05 by NorthernMan54