NorthernMan54/rtl_433_ESP #170: HI I have LILYGO® TTGO Meshtastic T-Beam V1.2 ESP32 LoRa 915MHz,

uriva9w, 2025-02-04:
### Current Situation

HI I have LILYGO® TTGO Meshtastic T-Beam V1.2 ESP32 LoRa 915MHz, what firmware do I have to select in OpenMQTTGateway ?

### Logs

```markdown
none
```

### Configuration

```json
None
```

### Environment

- OS: W11
- Software:
- Node:
- npm:

### Process Supervisor

not applicable

uriva9w, 2025-02-05:
I used lilygo_rtl433 but I just read 433 mHZ, I want to read R900 water meter. I've changed the RF configuration in WEbUI with no results. Please help me

NorthernMan54, 2025-02-07:
Hopefully someone in the community can assist with this, as I can only support devices I have personal access to.

For your device is it FSK or OOK?

uriva9w, 2025-02-07:
My device is FSK

ldina, 2025-02-28:
Each  T-Beam V1.2 ESP32 LoRa have a different version based on the frequency, if you bought a 915Mhz version you'll not be able to receive 433Mhz and vice versa

blaudp79, 2025-04-09:
> Each T-Beam V1.2 ESP32 LoRa have a different version based on the frequency, if you bought a 915Mhz version you'll not be able to receive 433Mhz and vice versa

They support wider frequency range though. The SX1276 for example has a frequency range of 137-1020Mhz and it is used on the 868 MHz and 915 MHz boards. If there is no further hardware filtering on the board, theoretically it should work on 433 MHz as well with a 433Mhz antenna.

sheilbronn, 2025-04-10:
Have you tried it in reality or is it an assumption?