NorthernMan54/rtl_433_ESP #172: Enable selection of specific protocol decoders

pauln, 2025-02-17:
### Current Situation

At present, either all decoders of the selected modulation type (OOK or FSK) are enabled, or `MY_DEVICES=true` limits it to a hardcoded subset of decoders.  This isn't always ideal, as other nearby devices which are supported but not the devices of interest can cause unwanted noise - especially when used in projects such as OpenMQTTGateway, where autodiscovery can result in excessive unwanted devices being discovered.

### Proposed Change

Add a way to select of a subset of protocol decoders to be enabled, similar to `rtl_433`'s `-R` flag .  This could be a compile-time option (as `MY_DEVICES` is), or a runtime initialisation option - which might make it easier to integrate into projects such as OpenMQTTGateway in a way that's configurable by the end-user.

### Additional Context

I have a range of 433MHz RF devices which I want to monitor, but a number of other devices are also being detected, some of which seem to be either badly behaved devices or not decoded correctly (such as a weather station which appears with varying IDs and wildly wrong measurements).  I'm using OpenMQTTGateway, and want to cut out this noise so that I can safely enable Home Assistant autodiscovery without a flood of irrelevant/useless devices being discovered by HA.

NorthernMan54, 2025-02-17:
Personally I run with auto discovery disabled to avoid this exact issue.  ( Actually I changed the discovery topic to something else, and manually copied over the discovery message for the devices I want )

I think that due to device noise issues and other challenges I don't think an effective code solution would be feasible, as the risk of phantom devices ie incorrect ID;s would always be there.   But disabling the discovery message is pretty effective.

pauln, 2025-02-18:
I've yet to see any phantom devices/IDs on any of the protocols I'm interested in - only my actual devices - so being able to limit it to just those protocols would seemingly cut out 100% of the noise in my case.

While manually copying discovery messages from a dummy topic into the "real" discovery topic is certainly one way to handle it, it'd be nice to not have to manually do things like that - especially for users who aren't as familiar with MQTT and/or who don't want to delve into it.

I do also wonder if limiting the enabled protocol decoders might help with reliability at all; I'm not familiar with the inner workings of either the original `rtl_433` project or this one, but it seems logical that focusing the decoding efforts on only desired protocols could reduce the odds of the wrong decoder being used for an ambiguous signal.