NorthernMan54/rtl_433_ESP #173: The website is down?

lachesis, 2025-03-05:
### Analysis

https://compatible.openmqttgateway.com/index.php/boards/

This page from the README has an HTTPS error then says:

```
Site not installed

The site is not yet installed. [[ configuration guide ](https://www.ovh.co.uk/g1585.configuration)]
If it has just been created, please wait 15 minutes.
```

### Expected Behavior

Site is up.

### Steps To Reproduce

Visit the site.

### Logs

```markdown
No logs.
```

### Configuration

```json
N/A
```

### Environment

Shouldn't matter

### Process Supervisor

not applicable

NorthernMan54, 2025-03-06:
This is an openmqttgateway issue, and not this package. Pls raise it there

closed 2025-03-06 by NorthernMan54

lachesis, 2025-03-06:
Oops my mistake.