NorthernMan54/rtl_433_ESP #175: Wiring CC1101 (E07-M1101D) to WT32-ETH01

diegoflyer, 2025-03-20:
### Current Situation

I searched a lot bu I found no data on how to wire the antenna to the esp, is there any infos?
Is there a place to map the GPIOS used?

### Logs

```markdown
no log
```

### Configuration

```json
wt32 + E07-M1101D
```

### Environment

- OS: home assistant
- Software:
- Node:
- npm:

### Process Supervisor

hb-service

NorthernMan54, 2025-03-21:
The current wiring of the gpio pins is done using the board definition.

diegoflyer, 2025-03-21:
Hi @NorthernMan54 I tried again, I built and upload the firmware, I was able to access the first time to set wifi but then no answer...
This is the code:

```
[platformio]
default_envs = esp32dev-multi_receiver-eth01

[env:esp32dev-multi_receiver-eth01]
platform = ${com.esp32_platform}
board = esp32-gateway
extra_scripts = ${com-esp32.extra_scripts}
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp32.lib_deps}
  ${libraries.rc-switch}
  ${libraries.smartrc-cc1101-driver-lib}
  ${libraries.rtl_433_ESP}
  ${libraries.newremoteswitch}  
build_flags =
  ${com-esp32.build_flags}
  '-DESP32_ETHERNET=true'
  '-DETH_PHY_ADDR=1'
  '-DETH_PHY_POWER=16'
  '-DETH_CLK_MODE=ETH_CLOCK_GPIO0_IN'
  '-DZgatewayRF="RF"'
  '-DZgatewayRF2="RF2"'
  '-DZgatewayRTL_433="RTL_433"'
  '-DZradioCC1101="CC1101"'
  '-DGateway_Name="OMG_ESP32_WT32-ETH01"'
  '-DvalueAsATopic=true'  ; MQTT topic includes model and device (rtl_433) or protocol and id (RF and PiLight)
; *** RF Module Options ***
  '-DRF_CC1101="CC1101"'  ; CC1101 Transceiver Module
  '-DRF_MODULE_CS=2'      ; pin to be used as chip select
  '-DRF_MODULE_GDO0=4'   ; CC1101 pin GDO0
  '-DRF_MODULE_GDO2=17'   ; CC1101 pin GDO2
  '-DRF_MODULE_MOSI=14'
  '-DRF_MODULE_MISO=15'
  '-DRF_MODULE_SCK=12'
;  '-DRF_MODULE_INIT_STATUS=true'    ; Display transceiver config during startup
custom_description = gateway using ethernet, need to be configured through PIO
custom_hardware = WT32
```

NorthernMan54, 2025-03-22:
Are you watching the serial output? it should log there any issues

diegoflyer, 2025-03-22:
@NorthernMan54 
```
08:42:19.244 -> 
08:42:19.244 -> 
08:42:19.244 -> 
08:42:19.244 -> ELF file SHA256: 770e9e42c1fdb40d
08:42:19.244 -> 
08:42:19.244 -> E (668) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
08:42:19.244 -> Rebooting...
08:42:19.244 -> ets Jun  8 2016 00:22:57
08:42:19.244 -> 
08:42:19.244 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
08:42:19.244 -> configsip: 0, SPIWP:0xee
08:42:19.244 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
08:42:19.244 -> mode:DIO, clock div:1
08:42:19.244 -> load:0x3fff0030,len:1184
08:42:19.244 -> load:0x40078000,len:13160
08:42:19.286 -> load:0x40080400,len:3036
08:42:19.286 -> entry 0x400805e4
08:42:19.855 -> Guru Meditation Error: Core  0 panic'ed (LoadStoreError). Exception was unhandled.
08:42:19.855 -> 
08:42:19.855 -> Core  0 register dump:
08:42:19.855 -> PC      : 0x401394d4  PS      : 0x00060930  A0      : 0x8013fe78  A1      : 0x3ffe3ad0  
08:42:19.855 -> A2      : 0x3f41ee28  A3      : 0x3f42833c  A4      : 0x0000000c  A5      : 0x00060f23  
08:42:19.855 -> A6      : 0x00060f20  A7      : 0x00000001  A8      : 0x801394d4  A9      : 0x3ffe3ab0  
08:42:19.887 -> A10     : 0x3ffbb26c  A11     : 0xfffc1fff  A12     : 0x00060f20  A13     : 0x00060f23  
08:42:19.887 -> A14     : 0x007bed4c  A15     : 0x003fffff  SAR     : 0x00000018  EXCCAUSE: 0x00000003  
08:42:19.887 -> EXCVADDR: 0x3f41ee28  LBEG    : 0x4008abe5  LEND    : 0x4008abf5  LCOUNT  : 0xfffffffe  
```

diegoflyer, 2025-03-22:
Hi @NorthernMan54 at the end I was able to build and install with this:
```

[platformio]
default_envs = esp32dev-multi_receiver-eth01

[env:esp32dev-multi_receiver-eth01]
platform = ${com.esp32_platform}
board = esp32-gateway
extra_scripts = ${com-esp32.extra_scripts}
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp32.lib_deps}
  ${libraries.rc-switch}
  ${libraries.smartrc-cc1101-driver-lib}
  ${libraries.rtl_433_ESP}
  ${libraries.newremoteswitch}
build_flags =
  ${com-esp32.build_flags}
  '-DESP32_ETHERNET=true'
  '-DETH_PHY_ADDR=1'
  '-DETH_PHY_POWER=16'
  '-DETH_CLK_MODE=ETH_CLOCK_GPIO0_IN'
  '-DZgatewayRF="RF"'
  '-DZgatewayRF2="RF2"'
  '-DZgatewayRTL_433="RTL_433"'
  '-DZradioCC1101="CC1101"'
  '-DGateway_Name="OMG_ESP32_WT32-ETH01"'
  '-DvalueAsATopic=true'  ; MQTT topic includes model and device (rtl_433) or protocol and id (RF and PiLight)
; *** RF Module Options ***
  '-DRF_CC1101="CC1101"'  ; CC1101 Transceiver Module
  '-DRF_MODULE_CS=2'      ; pin to be used as chip select
  '-DRF_MODULE_GDO0=4'   ; CC1101 pin GDO0
  '-DRF_MODULE_GDO2=17'   ; CC1101 pin GDO2
  '-DRF_MODULE_MOSI=14'
  '-DRF_MODULE_MISO=15'
  '-DRF_MODULE_SCK=12'
;  '-DRF_MODULE_INIT_STATUS=true'    ; Display transceiver config during startup
custom_description = gateway using ethernet, need to be configured through PIO
custom_hardware = WT32
```

NorthernMan54, 2025-03-27:
Tks for the update

closed 2025-03-27 by NorthernMan54