NorthernMan54/rtl_433_ESP #176: Looking for Example to decode Vevor 7in1

thomsenle, 2025-03-30:
### Current Situation

Dear all, 

first of all thanks for investing your time in such projects. 

I have a Vevor 7in1 Weather station at a location without Internet. As i unterstand it is possible with this Lib to decode the values send   over 868hz. Unfortunatly i dont know how to use this lib. Is there a example how to decode this FSK signal? My target is to send the decoded values via a second RF-Module via LoraWAN to TheThingNetwork. 

Thanks a lot for your helpf!

Thomas

### Logs

```markdown
none
```

### Configuration

```json
Esp32 with 2 SX127X modules
```

### Environment

- OS: 
- Software:
- Node:
- npm:

### Process Supervisor

hb-service

sheilbronn, 2025-03-30:
From looking at this weather station's description on Amazon it seems that this is an **wifi-connected weather station**:
So, you need a **wifi repeater** (instead of rtl_433_ESP) that you need bring into proximity of your weather station.
There are special outdoor wifi repeaters if you need that. Probably the most simply one will do since there are almost no bandwidth requirements. However, depending on your distances you might need special antennas or at least a repeater with a good wifi sensitivity. In this case you might want to follow recommendations for outdoor OpenWrt-compatible devices. That might give added flexibility if you need that later.

thomsenle, 2025-03-31:
The "wifi"-part is done with a esp"something" linving inside the display. The station itself is sending its data via 868mhz ( for europe ).
ANd this lib can use rtl_433 to decode. 

See here.

https://github.com/NorthernMan54/rtl_433_ESP/issues/150

I just need a example how to start. 
I tried with looking at the code of the open mqtt gateway, where it works. But its to complex for me.