NorthernMan54/rtl_433_ESP #177: Performance compared to full rtl_433 project

tpitman, 2025-04-28:
### Current Situation

I am using this project on a esp32 dev board connected to a CC1101 dev board. it works pretty good, but seems to miss a lot of the same packets that the rull rtl_433 project detects when using a Nooelec NESDR SMArt v5 device on my Mac or Raspberry Pi.

Is this expected behavior or is there something I might be able to improve this code base or get a better processor or is it because I am using the CC1101 and something else would work better?

### Logs

```markdown
none
```

### Configuration

```json
esp32 feather with DWEII CC1101 dev board
```

### Environment

- OS: none
- Software: none
- Node: none
- npm: none

### Process Supervisor

not applicable

NorthernMan54, 2025-04-29:
The reduced ability to receive packets is expected.  One solution is to place the device closer, and another is to place the antenna vertical.  I found vertical worked better.

tpitman, 2025-04-29:
Thank you.

Do you know why it doesn't work as well? Is it the slow processor, the CC1101 chip, the way you had to modify things to work with the esp32 or something else?

I ask with the hope that I can improved on the performance somehow.

tpitman, 2025-04-29:
This is excellent information. Thank you!

This seems to imply that the RTL USB dongle I am using with the RPi and the base rtl_433 project is a more flexible radio somehow.

I say that because I run that side by side with the esp32 / cc1101 and the rtl_433 on the RPi detects way more transmissions than the esp32 and cc1101.

I don't provide any special parameters to the RPi version, so it must be using the default values.

Does that imply what I think it does? The RTL is just a better radio at picking up these signals?

tpitman, 2025-04-29:
I am not sure I understand what the setBitrate testing is doing. I have 8 TPMS sensors near the CC1101. I have the antenna vertical. I set all 3 of those flags in the platform.io config file to be enabled. When I run the code and monitor the output in the serial window I see some output about starting at a bitrate and then finishing. I changed it from 60 seconds to 15 so I could cycle through it a little faster.

I then hit the U508 device for 10 seconds and then wait until the next Starting bitrate message comes out (about 15 seconds after I stop pressing the U508 button).

Here is the output from the first run:

```
N: Finished setBitrate:    1.00, count: 0
N: Starting setBitrate with    3.00
N: Finished setBitrate:    3.00, count: 0
N: Starting setBitrate with    5.00
N: Finished setBitrate:    5.00, count: 5
N: Starting setBitrate with    7.00
N: Finished setBitrate:    7.00, count: 10
N: Starting setBitrate with    9.00
N: Finished setBitrate:    9.00, count: 6
N: Starting setBitrate with   11.00
N: Finished setBitrate:   11.00, count: 10
N: Starting setBitrate with   13.00
N: Finished setBitrate:   13.00, count: 8
N: Starting setBitrate with   15.00
N: Finished setBitrate:   15.00, count: 5
N: Starting setBitrate with   17.00
N: Finished setBitrate:   17.00, count: 4
N: Starting setBitrate with   19.00
N: Finished setBitrate:   19.00, count: 3
N: Starting setBitrate with   21.00
N: Finished setBitrate:   21.00, count: 5
N: Starting setBitrate with   23.00
N: Finished setBitrate:   23.00, count: 3
N: Starting setBitrate with   25.00
N: Finished setBitrate:   25.00, count: 3
N: Starting setBitrate with   27.00
N: Finished setBitrate:   27.00, count: 6
N: Starting setBitrate with   29.00
N: Finished setBitrate:   29.00, count: 6
N: Starting setBitrate with   31.00
```

```
N: Finished setBitrate:    1.00, count: 12
N: Starting setBitrate with    3.00
N: Finished setBitrate:    3.00, count: 1
N: Starting setBitrate with    5.00
N: Finished setBitrate:    5.00, count: 4
N: Starting setBitrate with    7.00
N: Finished setBitrate:    7.00, count: 6
N: Starting setBitrate with    9.00
N: Finished setBitrate:    9.00, count: 13
N: Starting setBitrate with   11.00
N: Finished setBitrate:   11.00, count: 3
N: Starting setBitrate with   13.00
N: Finished setBitrate:   13.00, count: 5
```

I didn't realize that it had started with the bitrate step 1 for the first run, so I wasn't holding down the button. Not sure it matters because a third run didn't count very many for step 1 compared to the second run that counted 12.

There doesn't seem to be much connection between the 2 runs. The number of transmissions received just seems to be random.

I was hoping that I would see a curve around the best bitrate and be able to pick that rate.

I have to take a moment and thank you for engaging with me and all the work you have done on this. Thank you!

Any help you gave give is greatly appreciated.

NorthernMan54, 2025-04-29:
That is the methodology that can be followed for tuning, but you would need to review the CC1101 data sheet, determine which settings you want to try, then change the code to tweak the setting you want to try.

And you want to find the setting with the highest count.

But is it also picking up other signals, as they would also be included within the count ?

tpitman, 2025-04-29:
Ah, got it.

I just saw a comment in the code that says you grabbed the settings for the ookModulation of the CC1101 from lsatan.

Where did you get those and what kind of device was it for? Maybe that is the issue. Those settings are probably not a TPMS sensor.

NorthernMan54, 2025-04-29:
https://github.com/LSatan/SmartRC-CC1101-Driver-Lib

tpitman, 2025-04-29:
As I debug and look at the data I am getting I have a new piece to the puzzle that you might be able to help with.

I turned on this flag in the config file:

`  '-DRAW_SIGNAL_DEBUG=true'`

I triggered the tpms sensors to send.

I received one decoded packet, but also received 2 other things that start with `RAW (xxxxxxx)` where the x's are large numbers.

It would seem that there were supposed to be 3 signals, but something caused 2 of them to not get decoded.

I am hoping you can give me some pointers on how to take this raw data and figure out why 1 of the three decoded, but 2 didn't.

Here is all of the data:

1st RAW + decoded json:
```
rtl_433_ESP(6): RAW (188999): +116-88+138-115+115-115+140-94+139-94+150-94+138-92+138-114+115-116+142-92+142-100+142-92+138-114+115-115+140-94+143-90+152-108+138-110+111-107+132-101+159-83+138-111+138-110+111-112+128-104+132-108+138-111+138-111+110-108+133-100+130-111+138-110+138-111+110-111+130-105+131-108+138-110+138-111+111-108+264-209+134-108+248-109+134-232+132-110+232-216+151-105+248-218+137-110+132-108+139-110+111-110+138-111+136-105+130-104+135-111+110-111+138-110+136-106+228-231+116-115+139-95+137-120+128-109+138-110+137-80+132-102+160-82+139-110+138-111+110-110+260-108+137-208+258-99+130-247+110-110+130-104+266-206+139-92+138-116+231-121+116-244+266-217+136-107+128-106+236-235+121-126+136-110+248-112+103-236+137-110+138-110+233-225+237-244+246-243+109-109+138-111+233-123+136-221+242-108+136-110+111-239+136-65281+153-103+108-139+110-111+131-104+129-107+137-111+110-138+111-108+132-100+121-116+116-116+124-118+115-138+115-116+115-121+132-109+138-110+137-105+105-129+135-111+110-111+138-110+109-133+99-121+116-115+141-98+141-92+137-115+115-115+140-95+138-94+152-107+139-110+111-107+132-101+159-83+138-110+138-111+110-111+129-105+266-205+144-111+247-108+132-220+138-111+242-217+137-110+236-235+121-127+136-110+111-110+130-104+132-108+138-111+138-110+111-108+133-100+121-115+138-94+268-212+149-104+136-111+138-82+129-103+134-109+138-111+138-110+111-107+132-101+159-83+138-110+249-109+135-208+273-110+110-234+123-110+137-110+248-219+138-111+135-107+221-118+115-243+266-219+137-108+126-106+247-207+145-110+138-109+247-110+136-221+138-83+127-131+243-207+259-234+245-217+137-109+134-109+236-116+138-216+263-107+136-92+116-238+117-0 
Received message : 00:01:09 - {"model":"Schrader-EG53MA4","type":"TPMS","flags":"4c900080","id":"06C468","pressure_PSI":50.76321,"temperature_F":84,"mic":"CHECKSUM","protocol":"Schrader TPMS EG53MA4, PA66GF35","rssi":-47,"duration":188999}
```

2nd RAW (not decoded):
```
rtl_433_ESP(6): RAW (2626000): +138-83+138-110+137-103+130-105+140-107+110-111+248-109+134-231+232-116+138-217+132-104+135-111+138-111+245-109+135-107+136-84+137-111+138-209+124-118+131-110+138-111+246-109+135-215+136-111+138-83+260-97+121-232+142-93+140-92+272-215+274-217+137-110+248-109+110-111+138-110+136-106+124-216+136-109+248-218+89829-99+119-102+139-111+137-111+110-133+105-127+106-139+110-111+138-110+109-135+102-130+112-132+111-110+139-110+111-132+105-127+134-110+111-110+139-110+137-82+123-129+137-110+110-111+138-113+134-105+130-104+135-111+110-111+138-110+136-106+125-105+137-111+111-110+138-110+137-105+130-104+135-111+110-111+138-110+136-106+228-231+116-115+256-93+145-223+139-92+253-236+119-115+241-239+137-109+135-106+126-106+138-111+110-111+138-110+137-104+130-105+136-110+111-110+138-111+243-217+137-110+139-110+138-83+138-111+136-104+131-105+135-111+110-111+138-110+135-106+125-105+236-116+115-258+245-110+136-214+137-110+110-111+138-111+233-123+136-109+110-110+138-111+138-221+138-82+125-121+116-115+266-106+135-221+138-107+126-105+235-116+139-214+130-103+135-110+249-246+247-219+137-110+233-95+138-109+134-109+138-111+137-222+110-110+260-217+234-78758+138-92+139-115+121-130+137-83+138-110+133-105+128-107+138-110+138-83+138-111+135-103+130-107+136-110+138-83+138-111+134-105+128-106+138-111+138-83+141-107+137-103+131-105+135-110+139-83+138-110+135-106+126-106+138-110+111-110+139-110+137-104+131-104+135-110+111-110+138-111+136-106+124-105+138-111+233-214+153-103+247-109+135-233+109-109+236-237+148-104+246-244+109-109+132-109+138-110+111-110+138-111+136-103+130-106+135-111+110-111+138-110+134-106+230-232+116-138+136-105+107-138+111-110+139-110+110-110+129-104+132-108+138-111+138-110+111-107+264-104+130-219+247-108+134-231+135-110+138-110+111-109+238-128+127-94+115-115+139-116+115-254+107-109+119-116+123-118+258-99+131-247+109-110+129-104+267-109+137-208+121-118+117-115+256-214+268-214+137-109+238-135+109-111+138-110+138-111+111-236+126-98+271-217+246-57316+150-99+121-116+138-93+149-93+138-92+138-115+114-117+142-92+141-100+142-91+138-115+115-115+140-94+138-95+151-108+138-111+110-107+132-102+159-82+139-110+138-111+110-111+129-103+133-108+138-111+138-110+111-107+133-101+130-111+138-110+139-110+111-110+130-104+132-107+138-111+138-111+110-108+133-100+131-110+138-111+249-221+138-108+232-129+109-234+120-118+231-237+143-92+269-215+137-109+134-105+128-106+138-111+111-110+138-110+136-103+130-106+136-111+110-111+138-111+241-243+110-110+138-110+138-111+110-109+133-100+130-110+138-110+138-111+110-111+130-104+131-107+236-117+138-245+239-108+136-221+121-70+67-107+100-130+137-110+233-95+138-110+134-109+138-111+110-249+110-110+129-104+133-108+236-117+138-230+117-122+132-108+234-120+118-231+118-118+141-92+270-215+246-244+137-110+247-109+110-111+138-111+135-106+100-240+136-108+248-218+247-111844+132-102+160-83+138-111+138-110+138-82+129-103+135-109+138-111+138-110+112-107+133-100+131-111+138-110+1 [... cut ...] 48-111+136-230+135-110+233-214+151-105+247-245+109-109+133-108+139-110+111-110+138-111+136-104+131-105+135-111+110-111+138-110+135-106+222-235+117-120+117-115+138-114+115-115+140-95+138-94+151-108+138-111+110-107+133-101+131-138+221-109+137-220+273-109+109-248+109-110+138-110+137-105+238-106+132-109+138-111+110-107+133-231+135-110+138-111+110-109+238-129+128-210+141-94+140-92+272-107+135-208+142-92+138-115+230-237+237-243+130-107+248-108+134-101+131-111+138-111+137-209+121-120+232-238+236-0 
```

NorthernMan54, 2025-04-29:
Basic's on how this works

rtl_433 has device decoders, that input a byte stream, and determine if its a valid signal by running it thru the decoders.

rtl_433_ESP - Uses those decoders and passes what ever it receives to them

Sometimes rtl_433_ESP cuts off the front of a signal, or doesn't cut off a signal or other weird things and the decoder just discards it

When testing like this, I like to have a rtl_433 beside me as well, so I look at the output from both devices and see if the undecoded signals are just noise, or a failure to properly receive.

The raw output is actually a waveform of the received signal.  The numbers are the width of the pulse, and +/- indicate it is a up or down.  That is what the rtl_433 decoders expect.

If your device is using FSK, tuning would likely improve things greatly.  I had just tuned code base against my wh51 soil sensor, but other devices may need different settings.  One other option with FSK, is to leverage the onboard FSK decoder, provided you can determine the settings.   This approach would require coding a dedicated signal decoder to replace the rtl_433 decoder

tpitman, 2025-04-29:
Sadly my signal is all OOK.

I do have rtl_433 with a Nooelec dongle on it side by side and am seeing many more packets coming out on it than on the esp32 / cc1101 version. That is why I am pusuing this so much. I know I am missing packets.

I am also 90% sure the above examples are packets from the exact same sensors that sometimes work fine.

Do you have any thoughts on a tool that can consume those RAW + / - strings I posted and show the signals as pulses with timing? Maybe that will show me something that will stand out?

NorthernMan54, 2025-04-29:
I though this compiler option enabled the rtl_433 style output, and it generates a url that will leverage their tools - RTL_ANALYZER          ; Enable pulse stream analysis ( note is very resource intensive and will not work with other modules )
RTL_ANALYZE=##        ; Enable pulse stream analysis for decoder ##

tpitman, 2025-04-29:
Hmm. I get a bunch of linker errors for missing functions. I will poke around and see if I can find them somewhere. If you happen to know that would be great, but I don't expect you to. You have been so very helpful and I am super grateful for you even doing this project and publishing it. It got me way further along than I ever could have on my own.

Here are the linker errors in case you are bored ;-)

```
/Users/tpitman/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32_cc1101/libd6a/librtl_433_ESP.a(pulse_analyzer.c.o):(.literal.histogram_sum$constprop$4+0x0): undefined reference to `MAX'
/Users/tpitman/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32_cc1101/libd6a/librtl_433_ESP.a(pulse_analyzer.c.o):(.literal.histogram_sum$constprop$4+0x4): undefined reference to `MIN'
/Users/tpitman/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32_cc1101/libd6a/librtl_433_ESP.a(pulse_analyzer.c.o): in function `histogram_sum':
/Users/tpitman/Projects/Holley/HolleyFirmware/TPMS/rtl_433_ESP/src/rtl_433/pulse_analyzer.c:47: undefined reference to `MAX'
/Users/tpitman/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/tpitman/Projects/Holley/HolleyFirmware/TPMS/rtl_433_ESP/src/rtl_433/pulse_analyzer.c:51: undefined reference to `MIN'
/Users/tpitman/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/tpitman/Projects/Holley/HolleyFirmware/TPMS/rtl_433_ESP/src/rtl_433/pulse_analyzer.c:52: undefined reference to `MAX'
/Users/tpitman/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32_cc1101/libd6a/librtl_433_ESP.a(pulse_analyzer.c.o): in function `histogram_fuse_bins':
/Users/tpitman/Projects/Holley/HolleyFirmware/TPMS/rtl_433_ESP/src/rtl_433/pulse_analyzer.c:134: undefined reference to `MAX'
/Users/tpitman/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/tpitman/Projects/Holley/HolleyFirmware/TPMS/rtl_433_ESP/src/rtl_433/pulse_analyzer.c:139: undefined reference to `MIN'
/Users/tpitman/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: /Users/tpitman/Projects/Holley/HolleyFirmware/TPMS/rtl_433_ESP/src/rtl_433/pulse_analyzer.c:140: undefined reference to `MAX'
/Users/tpitman/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32_cc1101/libd6a/librtl_433_ESP.a(pulse_analyzer.c.o): in function `pulse_analyzer':
/Users/tpitman/Projects/Holley/HolleyFirmware/TPMS/rtl_433_ESP/src/rtl_433/pulse_analyzer.c:294: undefined reference to `MIN'
/Users/tpitman/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: .pio/build/esp32_cc1101/libd6a/librtl_433_ESP.a(pulse_analyzer.c.o): in function `histogram_sort_count':
/Users/tpitman/Projects/Holley/HolleyFirmware/TPMS/rtl_433_ESP/src/rtl_433/pulse_analyzer.c:116: undefined reference to `MIN'
```

NorthernMan54, 2025-04-29:
MAX and MIN are compiler directives, I believe for max int or something

Your environment must not have them ( I'm Mac with Xcode )