NorthernMan54/rtl_433_ESP #189: Trying to get a RFM95W module to work with rtl_433_esp

jbroll, 2025-09-21:
### Current Situation

Hi All,
I'm trying to get generic RFM95W 915Mhz radio module to work with rtl_433_esp.  I have what I thought would be the hard parts, firmware and wiring, working pretty quickly.  My esp32-s3 card boots up and I see the logs registering all the protocols.  Then I see the status register dump from the SX127x.  After that I just see some very low signal RSSI lines and no packets.  I'm using the OOK_Reciever example with the GPIOs defined for my module connections.  There is an Accurite 5n1 Weather station with fresh batteries in the same room with me.  I see the Accurite data on the weather station panel.

The package lists the module as: 

  RFM95W 915MHz w/SX1276 wireless transceiver module HopeRF Original.

Included platform IO config and some logs in the other input boxes.

Thanks!

### Logs

```markdown
----- SX127x Status -----
RegOpMode: 0x2c
RegPacketConfig1: 0x00
RegPacketConfig2: 0x00
RegBitrateMsb: 0x68
RegBitrateLsb: 0x2a
RegRxBw: 0x01
RegAfcBw: 0x02
-------------------------
RegOokPeak: 0x08
RegOokFix: 0x50
RegOokAvg: 0x12
-------------------------
RegLna: 0x20
RegRxConfig: 0x08
RegRssiConfig: 0x00
-------------------------
RegDioMapping1: 0x00
----- SX127x Status -----
...
rtl_433_ESP(7): Average RSSI Signal -91 dbm, adjusted RSSI Threshold -82, samples 50000
rtl_433_ESP(7): Average RSSI Signal -90 dbm, adjusted RSSI Threshold -81, samples 50000
rtl_433_ESP(7): Average RSSI Signal -90 dbm, adjusted RSSI Threshold -81, samples 50000
rtl_433_ESP(7): Average RSSI Signal -90 dbm, adjusted RSSI Threshold -81, samples 50000
...
```

### Configuration

```json
[platformio]
default_envs = esp32s3-generic
src_dir = .
boards_dir = boards

[env:esp32s3-generic]
platform = espressif32
board = esp32s3box
framework = arduino

build_flags =
    -D LED_BUILTIN=2               ; if you have an onboard LED
    -D ARDUINO_ESP32S3_DEV          ; ensure correct Arduino defines
  '-DLOG_LEVEL=LOG_LEVEL_TRACE'
  '-DONBOARD_LED=LED_BUILTIN'
; *** rtl_433_ESP Options ***
  '-DRF_MODULE_FREQUENCY=915.00'
  '-DOOK_MODULATION=1'           ; False = FSK, True = OOK
  '-DRF_SX1276="SX1276"'
  '-DRTL_DEBUG=4'
  '-DRTL_VERBOSE=58'
  '-DRAW_SIGNAL_DEBUG=true'
;  '-DMEMORY_DEBUG=true'
;  '-DDEMOD_DEBUG=true'
;  '-DMY_DEVICES=true'
;  '-DPUBLISH_UNPARSED=true'
;  '-DRSSI_THRESHOLD=12'
  '-DOOK_FIXED_THRESHOLD=0x50'
;  '-DAVERAGE_RSSI=5000'
;  '-DSIGNAL_RSSI=true'
; *** RF Module Pin Mapping for ESP32-S3-CAM ***
  '-DRF_MODULE_MISO=1'      ; SPI MISO
  '-DRF_MODULE_MOSI=42'     ; SPI MOSI
  '-DRF_MODULE_SCK=41'      ; SPI SCK
  '-DRF_MODULE_CS=40'       ; SX1276 NSS
  '-DRF_MODULE_RST=39'      ; SX1276 RESET
  '-DRF_MODULE_DIO0=38'     ; SX1276 DIO0 (IRQ)
  '-DRF_MODULE_DIO1=47'     ; SX1276 DIO1 (optional IRQ)
  '-DRF_MODULE_DIO2=21'     ; SX1276 DIO2 (optional IRQ)
  '-DRF_MODULE_INIT_STATUS=true'
```

### Environment

- OS:
- Software:
- Node:
- npm:

### Process Supervisor

hb-service

NorthernMan54, 2025-09-21:
The way the code works with the receiver module is by polling the rssi value, and when it drops it indicates a signal is being received, it then starts receiving the signal.  And when the rssi goes back up it indicates signal is complete and sends the pulses for decoding.

I don’t think your module is picking up anything.

Can you share a link to module you bought and the wiring diagram you used?

jbroll, 2025-09-21:
Aliexpress Google login seems to be down, so I don't have the product link.  I'll get it when I can if that will be helpful.

I agree that I'm not getting any signal.  I've tried 2 of the modules and the results are the same.  I have a 5n1 weather station here and assume it to be broadcasting?  I see the data from the station on its dedicated info panel, so it must be sending something?  I had the short loop antenna that you see on most modules connected, then I switched to a 3.2" quarter wave straight wire - was that a good idea?

The little module has these labels on the card that correspond to this diagram on the aliexpress site.  I mapped them to available GPIO and then to the defines needed in the platformio.ini file.  Since it looks like the firmware is talking to the module I think I did all that correctly.

[image]

[image]

closed 2026-05-11 by NorthernMan54